usetrmnl/trmnl-firmware #5: Beautify the WiFi setup portal

ryanckulp, 2024-07-02:
we currently use this project to provide users an easy WiFi handshake between their local network and the device:
https://github.com/tzapu/WiFiManager

[image]

let's migrate to this library and make a custom web portal that matchs our brand:
https://github.com/CDFER/Captive-Portal-ESP32

paw3lx, 2024-07-08:
I talked to @ryanckulp about that issue. I'm investigating what are the options. The proposed library (https://github.com/CDFER/Captive-Portal-ESP32) is actually a simple captive portal (basic webpage). It would require development of the WiFi configuration portal, managing of credentials etc. 
I'm looking for another libraries, e.g. https://github.com/ayushsharma82/ESPConnect which looks something that we need - simple wifi captive portal, but I don't like setup of the portal part. 

It might be that the simplest approach would be to not use any library and build custom WiFi portal, so we can style it as we want - IMO it also shouldn't require that much time to develop.

stevekennedyuk, 2024-07-08:
https://github.com/revk/ESP32-RevK

PR #16 (2024-07-31): Added custom WiFi Captive Portal
- Adds new custom  WiFi configuration portal. 
- Resolves #5  

Status:
- The PR is currently in draft mode as there are a few minor issues to fix
- Feedback on the implementation and any potential improvements is highly appreciated.
- I will add few screenshots either today or tomorrow, so you can see how the portal look like

ToDo:
- [x] Discuss what to do when user enters wrong wifi credentials
- [x] When user enters wrong credentials, then WiFi sometimes changes its SSID to esp_*** (validate if its a case on beta device)
- [ ] Complete testing on the beta d [... cut]
closed 2024-07-31 by MichaelXlivnenko