usetrmnl/trmnl-firmware #6: Support for BLE setup

ryanckulp, 2024-07-02:
the TRMNL PCB supports BLE, so we could make this an option for the WiFi credential handshake. requested by Noah in [public comments](https://www.kickstarter.com/projects/usetrmnl/trmnl-the-e-ink-display-for-your-favorite-apps-and-news/comments).

our RPi prototype implemented a rough [Web Bluetooth API](https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API) option, but we replaced this with WiFi / Active Portal.

[image]

gdifiore, 2025-03-10:
Working on this

gdifiore, 2025-03-14:
1. **Is there a preference for another web-based UI?**
    I have a prototype branch for app-based provisioning, it was easier to implement as a PoC. I do believe keeping a non-web-based option could be beneficial as well.
2. **How should we determine which WiFi provisioning method to use?**
    For example, should we have a dropdown in the device settings, or attempt BLE provisioning if the captive portal fails after three retries?

Also, for reference on increased binary size from added BLE libraries:
```
# Latest `main`
Checking size .pio\build\local\firmware.elf
...
RAM:   [===       ]  28.2% (used 92404 bytes from 327680 bytes)
Flash: [=======   ]  65.5% (used 1288164 bytes from 1966080 bytes)
# WIP BLE WiFi Provisioning (nimBLE library)
Checking size .pio\build\local\firmware.elf
...
RAM:   [===       ]  30.0% (used 98356 bytes from 327680 bytes)
Flash: [=======   ]  74.1% (used 1457082 bytes from 1966080 bytes)
```

Omeryl, 2025-03-14:
Oh yikes. 10% of the flash spent just for provisioning is kind of scary to me.

gdifiore, 2025-03-14:
Agreed, that's why I paused after getting it working. If someone knows a smaller library than `nimBLE` I can take a look, but it may be the smallest possible.

gdifiore, 2025-03-17:
@ryanckulp Thoughts/ideas on this?

stevekennedyuk, 2025-03-17:
simpleBLE? Or direct use of the ESP-IDF and remove the unnecessary bits

gdifiore, 2025-03-17:
[simpleBLE](https://github.com/espressif/arduino-esp32/tree/master/libraries/SimpleBLE) doesn't appear to support creating Bluetooth characteristics, which is critical to receiving data from from the input (either app or web page).

The minimum implementation with ESP-IDF brought the FW binary size up to around 94%(!) of flash. The linker should already "remove unnecessary bits" from the lib, so manually editing code or compiler flags wouldn't change the final binary size.

0PandaDEV, 2025-04-11:
Am I right that this would make it possible to use the trmnl completely without Wi-Fi all over BLE if it were to be implemented? Because that would be a game changer for me as I turn my Wi-Fi off overnight but still need the data.

ryanckulp, 2025-10-02:
as @gdifiore discovered, this kinda functionality would use too much memory, so closing for now.