usetrmnl/trmnl-firmware #18: WiFi portal UX improvements

ryanckulp, 2024-07-31:
great work on the new WiFi portal! much prettier.

today i set up [v1.3.0](https://github.com/usetrmnl/firmware/commit/7c5f4918167b7bd7c1afbbd8c53b03da33d71cc1) and have a few suggestions:

1. "Scanning networks, please wait..." was a bit slow; maybe we can animate this a bit (loader / spinner, or simply fade in/out the text)
2. when networks appeared in a list, there were duplicates (see below). my "house" network was shown 3x, perhaps this is each of my 3x mesh network access points? in any case let's consider deduping by SSID name and show only the strongest (highest RSSI value) of each SSID variant?
3. both of my TRMNL devices auto-updated to v1.3.0, so both were broadcasting a "TRMNL" SSID. it would be nice to hide our own networks to prevent users from "_connect to TRMNL Wifi... then connect to TRMNL Wifi_..." infinitee loop UX confusion

**duplicate SSIDs + TRMNL SSID**

[image]

MichaelXlivnenko, 2024-08-01:
> "Scanning networks, please wait..." was a bit slow; maybe we can animate this a bit (loader / spinner, or simply fade in/out the text)

Added loader animation(will available from next version)

> both of my TRMNL devices auto-updated to v1.3.0, so both were broadcasting a "TRMNL" SSID. it would be nice to hide our own networks to prevent users from "connect to TRMNL Wifi... then connect to TRMNL Wifi..." infinitee loop UX confusion

All TRMNLs will hidden in networks list(will available from next version)

> when networks appeared in a list, there were duplicates (see below). my "house" network was shown 3x, perhaps this is each of my 3x mesh network access points? in any case let's consider deduping by SSID name and show only the strongest (highest RSSI value) of each SSID variant?

In progress

stevekennedyuk, 2024-08-01:
On Thu, 1 Aug 2024 at 13:38, Michael Xlivnenko ***@***.***>
wrote:

> "Scanning networks, please wait..." was a bit slow; maybe we can animate
> this a bit (loader / spinner, or simply fade in/out the text)
>
> Added loader animation(will available from next version)
>
> both [...]
When the TRMNL SSID is generated I would use something like TRMNLABCD where
the last digits are the last 4 MAC address digits or similar.
Makes it easier to identify multiple devices.

> when networks appeared in a list, there were duplicates (see below). my
> "house" network was shown 3x, perhaps this is each of my 3x mesh network
> access points? in any case let's consider deduping by SSID name and show
> only the strongest (highest RSSI value) of each SSID variant?
>
> In progress
>

I would also have the above as I have a MESH system in my flat.

Steve

-- 
NetTek Ltd  UK mob +44 7775 755503
UK +44 20 3432 3735  /  US +1 (650) 423 1390
social id stevekennedyuk
Euro Tech News Blog http://eurotechnews. <http://eurotechnews.blogspot.com>
net

MichaelXlivnenko, 2024-08-01:
@ryanckulp, all fixed, you can test it with [1.3.3](https://github.com/usetrmnl/firmware/commit/d740a0f674aa58afd94bc74ad61f8950469baa05)

closed 2024-08-01 by MichaelXlivnenko