usetrmnl/trmnl-firmware #19: On Thu, 1 Aug 2024 at 13:38, Michael Xlivnenko ***@***.***>

MichaelXlivnenko, 2024-08-01:
On Thu, 1 Aug 2024 at 13:38, Michael Xlivnenko ***@***.***>
wrote:

> "Scanning networks, please wait..." was a bit slow; maybe we can animate
> this a bit (loader / spinner, or simply fade in/out the text)
>
> Added loader animation(will available from next version)
>
> both of my TRMNL devices auto-updated to v1.3.0, so both were broadcasting
> a "TRMNL" SSID. it would be nice to hide our own networks to prevent users
> from "connect to TRMNL Wifi... then connect to TRMNL Wifi..." infinitee
> loop UX confusion
>
> All TRMNLs will hidden in networks list(will available from next version)
>
When the TRMNL SSID is generated I would use something like TRMNLABCD where
the last digits are the last 4 MAC address digits or similar.
Makes it easier to identify multiple devices.

> when networks appeared in a list, there were duplicates (see below). my
> "house" network was shown 3x, perhaps this is each of my 3x mesh network
> access points? in any case let's consider deduping by SSID name and show
> only the strongest (highest RSSI value) of each SSID variant?
>
> In progress
>

I would also have the above as I have a MESH system in my flat.

Steve

-- 
NetTek Ltd  UK mob +44 7775 755503
UK +44 20 3432 3735  /  US +1 (650) 423 1390
social id stevekennedyuk
Euro Tech News Blog http://eurotechnews. <http://eurotechnews.blogspot.com>
net

_Originally posted by @stevekennedyuk in https://github.com/usetrmnl/firmware/issues/18#issuecomment-2262948476_

yankobogdan, 2024-08-01:
If we can show it on EPD like that to match the WiFI AP name
That will make the identification feature reasonable
@ryanckulp What do you think?
[image]

ryanckulp, 2024-08-01:
IMO it's not necessarily to suffix a Mac address or random string. most people will not set up 2x at once, they will set up back to back. so we'd have a less sexy experience a lot of times, for no benefit.

on this note though, what if when we broadcast the signal we determine on the fly whether another TRMNL ssid exists, and if so we append chars + show on the EPD? :)

MichaelXlivnenko, 2024-08-01:
@ryanckulp, sure, I understand your point. However, it's important to consider that network scanning typically takes at least 3-5 seconds, if not more. This means you can expect an increase in the response time of the device. It's important to note that if the devices are not connected, they won't have a friendly ID, which is the main visible key. Also, sharing the Mac address is not advisable as it's not public. In this scenario, there would be nothing to display on the screen and no network name prefix.

yankobogdan, 2024-09-06:
Doesn't seems relevant for now