usetrmnl/trmnl-firmware #20: Event handler when the mac address is not added to the system

MichaelXlivnenko, 2024-08-02:
Currently, there is no event handler when the MAC address of the device is not added to the system. The device receives the appropriate status code and performs a reset. As a result, the user constantly sees the message that Wi-Fi is not saved. I think the best solution would be to add a screen similar to when the device is not linked to the server using a friendly ID. However, this screen should indicate the MAC address and provide a link to the help section https://help.usetrmnl.com/en/

ryanckulp, 2024-08-02:
good idea. highly unlikely this will happen because we test that the WiFi signal is broadcasting on every device before shipping, but at scale we could miss something during assembly.

same logo screen as Friendly ID, new message:
"Your device was not found. Email team@usetrmnl.com this Mac Address: XX:XX:XX:XX:XX"

ryanckulp, 2024-09-29:
we now print a Serial number (ex: "TRMNL_02115") on all PCBA and these are also stored in db -> Devices table, so i think we're OK here.

closed 2024-09-29 by ryanckulp