usetrmnl/trmnl-firmware #46: Feature Ask: Configurable Sleep Schedule

JackDDavis, 2024-11-15:
Would like the ability to configure sleep schedule. The default option for sleep schedule is 10p-8a though this does not always align to a user's schedule. This may prompt more users to enable thereby reducing the amount of pinging across all devices while extending battery life for end user TRMNL devices.

ryanckulp, 2024-11-15:
thanks Jack. we will get this done at the web app (vs firmware) layer, but i'll keep this issue open until it's live.

JackDDavis, 2024-11-15:
Great news. Question: Is there a better place/separate repo for submitting/tracking web app specific asks?

ronakjain90, 2024-11-16:
Hi @JackDDavis - This is a good place to start: https://usetrmnl.com/roadmap

ryanckulp, 2024-11-16:
@JackDDavis custom sleep schedules are now live, cheers.

[image]

closed 2024-11-16 by ryanckulp

JackDDavis, 2024-11-17:
Hey Ryan. In testing alternative sleep schedules, it works successfully though the display still lists 8am during sleep