usetrmnl/trmnl-firmware #48: More fun welcome screen

ryanckulp, 2024-11-30:
since v1.4.3 we use the TRMNL logo vs a 'snow' effect on boot. but it might be more fun to have a few welcome phrases instead.

Hello
안녕하세요
Bonjour
Hallo

GoReNeY, 2024-12-06:
That's a nice idea, but I don't think we have fonts for languages which are not english in firmware, also I think that the render on the device would not be neat enough. 

But, I'd like to clarify that the TRMNL logo displayed now is an .bmp image, and I used it just because we already have it in device memory :)
But we're free to add any other picture to use instead

ryanckulp, 2024-12-06:
cc @loucyan if you wanna show some welcome message beyond our logo, make a 48kb BMP with everything combined and we can swap it out.

closed 2025-02-20 by ryanckulp