usetrmnl/trmnl-firmware #49: User request - fall back to previous render if new fetch fails

ryanckulp, 2024-12-04:
customer reports that their company WiFi failed, so when TRMNL fetched new content it rendered the error screen.

they asked if it would be possible for our PCB to keep the former screen in memory _until_ JSON response with valid (includes https) `image_url` is returned. not sure if this is feasible but wanted to share.

GoReNeY, 2024-12-06:
We already have the last shown picture saved in the flash, so it would not be a big deal, but I'd like to clarify, how exactly this feature has to be implemented? Like, for all users, or maybe some fancy "I don't need errors, show the last used picture instead" checkbox?

ryanckulp, 2024-12-06:
good question. it could be a single toggle, like "soft fail / ignore errors." then the device would never show error screens, just stale / previous content instead.

but to be clear, the user was simply suggesting to not show an error **if the local network is not available**. however perhaps FW can't be sure if it is our device* that failed to connect, vs if a local network itself is down.

GoReNeY, 2024-12-06:
Oh, I understood, will look what we can do, thx for clarifying!

GoReNeY, 2024-12-06:
Okay, I did some research, and what I figured out:

We can tell if local network is down, by scanning all networks before performing connection, but that scan will take significant time, and it will take time for all bootups and all users

maybe can be implemented as optional feature

ryanckulp, 2024-12-06:
sounds like a nice approach. making it optional also great. we can add this as bool flag to our `api/display` payload response as `ignore_network_down` or similar, let me know what's best.

presstoexit, 2025-03-28:
I'm not the person who asked for this in the first place, but to my mind a good solution would be to overlay a banner that says something like "no network connection on last refresh", or "stale data - last updated YYYY-MM-DD HH:MM".  Much less effort than doing a scan of all the networks each time, no need for an option.

(A similar solution might be useful when the data _provider_ is down; i.e. the device can reach the TRMNL server, but the last time the server polled the data provider for the plugin it couldn't be reached or returned an error; rather than just display an error message, it _might_ be useful to keep displaying the last valid bitmap, with an overlay to note that it's stale. But it's harder to know if that's a good or bad idea given the variety of uses of different plug-ins.)

ryanckulp, 2025-10-02:
generally speaking, stability has improved a lot over the last 11 months since posting this issue. we also have retries + exponential backoff, better troubleshooting tips for maximizing WiFi compatibility, and a fallback solution that saves the previously displayed image in memory.

thinking we can close this as "not planned," however @donatj that is a bummer to hear you need to force-skip. my main concern remains: we don't want a soft fail strategy to hide a flaky wifi connection. do you agree that making this an opt-in flag, such as `ignore_network_failure` true/false is the most suitable approach?

i can make that happen on the core side, where it becomes a Device settings toggle. then we can PR here to use that flag in the HTTP error handler to show the previous image vs an error message. (but again, this doesn't improve your connection, so you may see stale content)

jasonwaters, 2026-08-17:
this is seriously ruining my experience with TRMNL, all my devices now regularly show this screen and require manual intervention.  likely just because of a temporary wifi hiccup.  huge downgrade in user experience!!

BTW, my wifi isn't shoddy.  actually it's excellent, i have unifi network access points.  if anything this is due to the device having too many access point options to connect to and it's failing somehow.  🧐

[image]