usetrmnl/trmnl-firmware #50: Option to keep last image in place while sleeping

tcfurrer, 2024-12-10:
Currently while sleeping my display is going to an all black image with "z z z" and "...mrmphmp".  At least for some plugins, it makes more sense to just leave the last image in place.  Please add an option for this.

If it's important to be able to tell whether the device is sleeping, how about leaving the last image but with a superimposed small sleep icon in the corner?

Just to give one concrete usage model example: I'm using the "Price" plugin to display stock prices.  While the market is closed, I'd really like the device to sleep while still displaying the closing price.

GoReNeY, 2024-12-11:
Hi, thanks for your issue, but I think it's more web related

@ryanckulp, could you answer there?

arnebr, 2024-12-21:
That would be awesome, maybe choose a layout in the trmnl setting like showing a topbar or a bottom footer

ryanckulp, 2025-02-07:
this feature has been deployed, just toggle off Sleep Screen from your device edit view.

closed 2025-02-07 by ryanckulp