usetrmnl/trmnl-firmware #51: Provide more information to the screen along with error messages

tcfurrer, 2024-12-10:
The error screens for firmware update failures, wifi connectivity failures, API connection failures, and image content malformed failures, etc all just give a short description of the error type, but don't provide additional useful details.  It would be nice for these error screens to include relevant information such as the exact URL which couldn't be connected, whether DNS resolution failed VS socket connection failed, what IP address the DNS resolution returned, how many bytes were downloaded when image content is malformed, etc.  At least this sort of information would be nice for _developers_ to have directly on the screen when problems happen...

tcfurrer, 2024-12-11:
I agree those logs are much better for the cases where they are available. But what about the specific subset of errors that prevent the device from uploading? Is there some existing way to get more information directly from the device in those cases?

GoReNeY, 2024-12-13:
Currently, in our firmware we have more detailed logs that goes through serial port (USB) for debugging purposes, you can build firmware by your own following the guide in README of repository to see them

closed 2024-12-20 by GoReNeY