usetrmnl/trmnl-firmware #53: Feature Request: Extended Sleep settings

Belgabor, 2025-01-11:
The sleep setting is quite nice, but it could imo use more fine-grained options.
- Two sleep cycles per day (eg for sleep and work hours)
- Separate configuration per day of the week (eg different schedules for workdays/weekend)
- Ability to pin a certain day (ie always use that days schedule until turned off). Use would be eg to pin a weekend schedule while on holiday.

yankobogdan, 2025-01-11:
@Belgabor Thanks for your feedback
That makes sense to me
In TRMNL architecture sleep duration is dictated by the server

@ryanckulp could you consider this feature request for Web APP?

ryanckulp, 2025-01-11:
thanks @Belgabor for the suggestions. we recently refactored Sleep Mode to support intra-day sleep (ie 14:00-19:00), so we'd like to make sure that's stable before extending. but extracting sleep hours from device level attributes to a `sleep_schedules` table with options for days of week, multiple times of day, etc rules sounds neat. 

stay tuned!

ryanckulp, 2025-11-02:
closing this because the new [Scheduler](https://help.usetrmnl.com/en/articles/11663305-playlist-scheduler) and [this guide](https://help.usetrmnl.com/en/articles/12387950-create-a-custom-sleep-screen) offer alternative ways to spoof intra-day sleep behavior.