usetrmnl/trmnl-firmware #55: Inkplate Support

leoherzog, 2025-02-01:
[The Inkplate devices by Soldered](https://soldered.com/categories/inkplate/) (particularly the 9.7" 1200×825px [Inkplate 10](https://www.crowdsupply.com/soldered/inkplate-10)) would be fantastic TRMNL devices. I've been using one in a picture frame with some homegrown Apps Script code for a few years now for family calendaring, but getting it pulling from TRMNL would tons more features and flexibility than my current janky code.

itopaloglu83, 2025-02-01:
I think that's a great idea for folks who want to bring their own devices. It's a custom PCB with everything on board and it makes things a lot easier for newcomers. 

I bought a 6-inch one yesterday to experiment. Though, the TRMNL firmware is device specific so it might be easier to just interact with the TRMNL web api directly. Most of the device functionality comes from the web api and the 60+ plugins.

yankobogdan, 2025-02-03:
@leoherzog I don't think that we will work on the integration of 3-rd party HW, we just don't have time for that and focusing on supporting our HW

But feel free to contribute by adapting your existing code to use TRMNL API or even better, - make TRMNL FW run on another HW

Inkplate 10 is based on ESP32 while TRMNL OG is based on ESP32-C3 and both support the Arduino framework
Consider that the HW interface for EPDs is completely different

closed 2025-02-03 by yankobogdan

leoherzog, 2025-02-03:
Thanks for your thoughts. I suppose this really should be an issue in the TRMNL server software that says "enable custom screen resolutions" :)

jesserockz, 2025-02-03:
Using ESPHome (which supports inkplate diplays) You can technically pull the images from TRMNL cloud provided you have a valid TRMNL mac address/id/api key

https://github.com/jesserockz/esphome-trmnl/pull/6

lanrat, 2025-07-01:
I recently added Trmnl support to [Homeplate](https://github.com/lanrat/homeplate), my firmware for the Inkplate 10.

I initially tried using ESPHome, but could not get ESPHome to handle the larger png files for the Inkplate 10's screen resolution with the online_image component. It would often run out or memory, crash, or hit a watchdog timer. Smaller lower-resolution images worked ok.