usetrmnl/trmnl-firmware #56: Deep Sleep Power Efficiency

itopaloglu83, 2025-02-01:
The deep sleep power consumption can be further reduced another `33uA` by disabling some of the internal pull-up/down resistors. Some comments event claim a total power consumption of `7uA` only compared to current value of `100uA`. I don't have a power profiler so it's not possible for me to reliably measure what the changes would be.

Could someone who has a power profile try disabling the pull-up/down resistors and share their results? More info below.

https://electronics.stackexchange.com/a/530152
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html#configuring-ios-deep-sleep-only

Practically speaking all we need to do is to isolate some pins by calling `rtc_gpio_isolate(GPIO_NUM_12);` before sleep. There are varying comments on the internet about which pins to disable and espressif documentation says it depends on the module and which pins are directly connected to power.

yankobogdan, 2025-02-02:
Hi @itopaloglu83 

Thanks for your comment on possible improvement
I completely agree that this approach could help us to save some uA for pins of ESP32-C3 which are connected to external peripherals and what's interesting even floated with ESP32 internal resistors.

JFYK
Around half of the power consumption while the device in RTC sleep is a DC-DC IC quiescent current, also we have a resistor divider for battery measurement. Only around 30uA of consumption, it's an ESP32-C3 itself and the EPD display. 

For the next version of TRMNL, we are going to use a more advanced battery (%) measurement IC and significantly more effective in terms of quiescent current DC-DC

We will give it a try and report back with the results

Stay tuned!

itopaloglu83, 2025-02-03:
Awesome! It sounds like it's going to be event better.

Please feel free to close the issue or leave it open for the curious among us to give it a try. We might squeeze a couple more days of battery life from the existing hardware.

closed 2025-10-21 by Fr3dr1ckson