usetrmnl/trmnl-firmware #57: How to build from command line?

coolaj86, 2025-02-03:
I see that this is a cpp project.

I was expecting there might be a Makefile, or CMake, or just a gcc or llvm line, or something in the GitHub Actions, but I don't see anything that looks like I could run it and get the firmware.

Is there a way to build this locally, headless, with standard(ish) cpp tools?

tomas-bubenicek-fr, 2025-02-03:
The project uses PlatformIO as a build system, you should be able to build it using `pio run` with [PlatformIO Core](https://platformio.org/install/cli), thoug it probably would be useful to have this mentioned in the Readme file

coolaj86, 2025-02-03:
Is that to so it can only be built in platformio's cloud?

It can't be built locally?

ryanckulp, 2025-02-03:
@coolaj86 you can build locally with the PlatformIO plugin, just click the "Build" icon and then the "Upload" icon when you're ready to flash. detailed instructions are on the README.

to demonstrate non reliance on the cloud i just disabled my network and it built in 6 seconds:

[image]

or as @tomas-bubenicek-fr suggested, `pio run` should also do the trick.

closed 2025-02-03 by ryanckulp

tomas-bubenicek-fr, 2025-02-03:
The "without any dependencies to host machine" just means you can build it on any device that runs python btw. PlatformIO don't even provide their own cloud services.

coolaj86, 2025-02-20:
@tomas-bubenicek-fr What's the command to write the firmware to the device?

Since it also works in a web browser, I'm imagining it's a very simple, standard CLI serial tool?

And perhaps `pio` provides some abstraction as well? I was looking through the docs but didn't find a section on how to write the firmware.

tomas-bubenicek-fr, 2025-02-20:
I don't have the device on hand, but shouldn't `pio run --target upload` be enough?

coolaj86, 2025-02-20:
I don't have the device either, but that's helpful. I was searching for "firmware" and something like a `flash` or `write` command in both sets of documentation and I wasn't finding anything.

I wouldn't think of searching for `upload` - that's how I think of photos and mp3s, not device firmware. Is that colloquially how people in the industry refer to it? Or just a weird choice by `pio`?