usetrmnl/trmnl-firmware #58: Note: Easy HTTPS for local networks (use Caddy + DuckDNS or similar)

coolaj86, 2025-02-03:
This is probably better suited for notes in the BYOS section, but I see a number of issues here have referenced the issue of running HTTP rather than HTTPS at home, and I'm a big believer in "all networks are adversarial, just use HTTPS whenever you can".

And since Let's Encrypt came out several years ago, it's now trivial to use HTTPS - even at home, even on a non-public network - without having to bypass any security measures.

In generic terms:

1. Get a (free) domain, and point it to the local ip address (such as `192.168.0.10`)
2. Use a tool that supports that DNS provider to get free HTTPS certs
3. Configure a webserver to use those certs, and proxy to the API server

More specifically:

- You can get a free domain with DuckDNS (like noip, afraid, and others)
- Let's Encrypt provides free HTTPS certs that work in all browsers (Brave, Safari, etc)
- Caddy is a webserver that supports Let's Encrypt  via HTTP or DNS
- Caddy can reverse proxy to the Ruby web server

Technically:

1. You download `caddy` with DuckDNS support from 
<https://caddyserver.com/download?package=github.com%2Fcaddy-dns%2Fduckdns> or build it with `xcaddy` (see <https://webinstall.dev/xcaddy> for notes).
2. You get your DuckDNS API token and stick it in `caddy.env`:
   ```sh
   MY_DUCKDNS_TOKEN=xxxxxxxx-xxxx-4xxx-8xxx-xxxxxxxxxxxx
   ```
3. You create a `Caddyfile` for your duckdns domain:
   ```caddyfile
   example.duckdns.org {
       tls {
           dns duckdns {env.MY_DUCKDNS_TOKEN}
       }

       reverse_proxy localhost:3000 # change to the BYOS port
   }
   ```
4. Run the HTTPS server
   ```sh
   caddy --envfile ./caddy.env --config ./Caddyfile --adapter caddyfile
   ```

## Notes

### Add caddy as a System Service

`serviceman` is a simple tool written in shell script that templates out the appropriate system launcher config file for launchctl (macOS), systemd (most Linuxes), or OpenRC (Alpine and similar Linuxes).

```sh
curl https://webi.sh/serviceman | sh
. ~/.config/envman/PATH.env

serviceman add --name caddy -- \
    caddy --envfile ./caddy.env --config ./Caddyfile --adapter caddyfile
```

You can, of course, find articles on how to set up launchctl or systemd or OpenRC individually, but it's one of those things that can be very tedious and error-prone, has a lot of nuance, and for many people tends to be more of a barrier to entry than a particularly useful skill to acquire.

### Build caddy with DuckDNS support

As you'll see on the page, sometimes the online build server is overloaded or otherwise fails, but building it yourself is very easy:

```sh
curl https://webi.sh/go | sh
curl https://webi.sh/xcaddy | sh
. ~/.config/envman/PATH.env

CGO_ENABLED=0 xcaddy build \
    --with github.com/caddy-dns/duckdns \
    --output ./caddy
```

kyso, 2025-02-03:
Right now the firmware does not even validate TLS certificates.

coolaj86, 2025-02-03:
Yeah, and if I'm understanding the limitations of the esp32, that probably won't be a priority.

However, the demo BYOS is also accessed by web browsers, and things like password managers and just hitting the dashboard work much better with the real certs.

It's possible to add certs to the local OS's chain, but... [bad things sometimes happen](https://www.eff.org/deeplinks/2015/02/further-evidence-lenovo-breaking-https-security-its-laptops) when people do that...