usetrmnl/trmnl-firmware #59: What prevents brute-forcing all MAC addresses in the production database?

coolaj86, 2025-02-03:
I haven't bought this yet, and I haven't found a setup video showing the initial setup process, but from what I saw in the BYOS demo, and what I see in the README here, it looks like I could brute force every single possible MAC address within a matter of a few months slowly enough to get every single API Key without detection.

(once an attacker knows the first 3 bytes for the manufacturer designation, there are only ~16 million possible addresses, which a bot could try every few seconds from a a few $1/month rotating IP addresses and get them all relatively quickly without any undue burden on the server - and orders of magnitude faster if there's a pattern to the addresses)

- Are the MAC addresses registered?
- Will the production API key return every time I supply the MAC address to `/api/setup/`?
- What happens if a bot registers the MAC address before I do?
- What prevents a bot from crawling to get every customer's API key?

jesserockz, 2025-02-03:
You really should not have made this a public issue, but privately disclosed this directly to TRMNL and given them a chance to fix it. 

Very bad security practise

coolaj86, 2025-02-03:
@jesserockz I understand your concern, but I don't see this as something sophisticated that requires specialized knowledge or advanced techniques.

There aren't very many customers, devices are not actively shipping, and I have no proof of concept or concrete evidence of a vulnerability.

I don't have access to the API yet, but I'm seeing the code that everyone else can see, and I'm raising questions based on what I see in the README and the code immediately surrounding what we're instructed to change in the README.

ryanckulp, 2025-02-04:
we've deployed rate limiting on the core server. a Mac ID enumeration attack is not feasible, but you are welcome to try and we'll pay a bounty if you find something.

next, we set `device.setup_at=<timestamp>` upon the first successful ping to `api/setup`. this allows us to disable subsequent requests for that device's API key.

however a device can be soft reset by its owner at any time, client-side (holding BOOT button). so we're considering a solution like this:

```
if @device && !@device.setup?
  # return api key
elsif @device
  # { message: "Device already paired, please un-pair inside your TRMNL account" }
end
```

but it's still not perfect, because it requires a user to log into TRMNL. what if the device was sold, swapped, given away, recycled? the new owner would be stuck. so we're still riffing, and you are welcome to propose another approach.

coolaj86, 2025-02-04:
> we've deployed rate limiting on the core server. a Mac ID enumeration attack is not feasible

That sounds pretty good to me. 👍 

> we set `device.setup_at=<timestamp>` upon the first successful ping to api/setup`

Same as the `activated` thing I was mentioning above. 👍 

> what if the device was sold, swapped, given away ...

I would expect the exceptional cases to be handled via human support. \
As long as you do better than Apple (which is to say "not actively fighting against reuse"), you're doing great in my book!

> you are welcome to propose another approach

I just wanted to know that it was considered and handled because the entropy of 3 bytes of MAC is very low.

More ideas than a proposal, but there are two rate limit strategies that I like in particular:
- exponential backoff - if an IP gets a non-200 on an endpoint that is attractive to bots (auth, etc), a counter is kept and an artificial sleep is introduced for some number of seconds
- always have a constant delay (regardless of db time or success or failure), but just send back phony data after failure \
  (i.e. it either looks like an error, or looks like a success, but the data is nonsense)

In the case of this device, I don't think it should be possible for a request to be incorrect in production - and if it is, it won't fix itself without a support call or a re-flash, or whatever - so it could be a straight 15 minute block per IP per bad request. If a person has a malicious device on their network phishing for their device info, and is also a legitimate customer... they must be very important and the fine people outside are probably about to pop in, put masks over their heads, and take them to a place where notifications are the least of their worries.

I'm glad to know the mitigation strategy.

Thanks!

closed 2025-02-04 by coolaj86