usetrmnl/trmnl-firmware #61: Option to choose self hosted server at initial login

kaiwei, 2025-02-05:
Instead of forking the firmware to change the base url, is it possible (via the UI) to choose and the input a self hosted server address - as an option? As such, the default firmware will always work for self hosting or the default cloud server out of the box. 

Something like this below:

https://images.app.goo.gl/xVb1W35YxUTC9QZUA

Omeryl, 2025-02-05:
I was actually going to work on this, once I got my physical device. Is there any appetite for this @ryanckulp? I’m fine with it even being a hidden page not linked from the default captive portal to make it so your main customers are not led astray!

gitstua, 2025-02-05:
This would be great @ryanckulp - we could also point at a preview cloud instance to help test new updates. As the userbase grows we want to figure out how to keep any markup changes stable for standard users and let dev hackers use a preview server.

ryanckulp, 2025-02-07:
thanks everyone, i really like this idea. for context, i initially wanted to provide a firmware config GUI here:
https://usetrmnl.com/flash

then users could mod multiple settings and flash directly to their device without installing any dependencies.

but the ESP Web Tools lib does not actually build files (AFAIK), just facilitates their upload to a peripheral. so it probably isn't worth running server-side builds and opening up that can of worms.

ergo! FW maintainer @GoReNeY -- any objections to making `API_BASE_URL` definable by users inside the WiFi captive portal? also tagging @paw3lx who did most of the [relevant work](https://github.com/usetrmnl/firmware/pulls?q=is%3Apr+is%3Aclosed+author%3Apaw3lx).

Omeryl, 2025-02-08:
Cool, I've been working on this and I'm nearly complete, I'm just testing it now and will PR it later today hopefully.

ryanckulp, 2025-02-20:
completed via #66