usetrmnl/trmnl-firmware #67: Send color bit depth together with width and height

danielsitnik, 2025-02-12:
I see that pull request https://github.com/usetrmnl/firmware/pull/60 added support for the firmware to send the device's width and height.

I believe the purpose of this is to support different sized devices in the future, and so the server (and BYOS implementations) can adapt to that without the need to have these values stored on a database.

So in the same spirit, shoudn't the device color depth be sent as well?

This influences the screen rendering process just like width and height. And I remember @ryanckulp saying that a future device would support 4bit images.

clach04, 2025-02-22:
I'm defintely interested in this. I'm waiting for my device so in the mean time  I'm experimenting with a smaller screen and 4-bit color (and a super simple micropython client).

delhoume, 2025-04-28:
I think it may be more appropriate to send not exactly the bit depth, but also what palette  is needed.

A simple example is that 1 bit dfor a given device do  not tell what is 0 and 1 value mapped to.

So we need a palette to be added.  That will allow more flexiibility.

Having the palette  for the device available, the server can generate a compatible bitmap  copying .
Thje devjce can reorder the palette entries is necessary

Instead of text, it would be so nice to send a compatible BMP such as this

[image]

for 2 bit per pixel, a minimal BMP is 54 (header) + 4 * 4 (colormap) + 4 (2 byte needed but BMP pads lines to 32  bit) 
= 66 bytes

danielsitnik, 2025-05-05:
Let's not overcomplicate things... this about sending the bit depth as a HTTP header, and then the server can map it to the actual palette.

delhoume, 2025-05-05:
> Let's not overcomplicate things... this about sending the bit depth as a HTTP header, and then the server can map it to the actual palette.

bit depth can be the same for different color panels, as it can mean black white 33% gray 66% gray or black white red yellow (o mqaybe even black white yellow red), this is  why I suggested to send the palette.

Maybe 4 bit means always gray levels but I am not sure.

Sending the palette will not make the request much larger, with 3 bytes per color entry

ryanckulp, 2025-11-02:
to @delhoume 's point we recently added a new paradigm on core, `appearances`, which allows flexible combinations of attributes like color palette, bit depth, etc. so while `width` and `height` may be useful, bit depth would not be. and re width/height, doesn't [this](https://github.com/usetrmnl/trmnl-firmware/issues/234#issuecomment-3478033071) solve for that as well?