usetrmnl/trmnl-firmware #71: Multiple WiFi network credentials

ryanckulp, 2025-02-24:
i thought we already supported this, but maybe not.

some users transport their TRMNL between work <> home and expect it to automatically connect to the appropriate network, like other devices.

let's refactor [wifi credentials](https://github.com/search?q=repo%3Ausetrmnl%2Ffirmware%20ssid&type=code) to an array/collection that cycles each known network pairing before giving up and exposing the captive portal. and on boot, the last connected network should be attempted first before cycling.

michaelbolanos, 2025-02-24:
Here's a Python script that manages Wi-Fi connections by cycling through known networks before exposing a captive portal. It attempts to reconnect to the last successful network first. This is designed for Linux-based systems using nmcli (NetworkManager CLI).

Features:
Stores Wi-Fi credentials in an array.
Attempts to reconnect to the last successful network first.
Cycles through all known networks if the first attempt fails.
If all connections fail, it exposes a captive portal.

[wifi-refactor-rename-to-py.txt](https://github.com/user-attachments/files/18945874/wifi-refactor-rename-to-py.txt)

Omeryl, 2025-02-24:
@michaelbolanos I'm rather confused how that is relevant here.

michaelbolanos, 2025-02-24:
> [@michaelbolanos](https://github.com/michaelbolanos) I'm rather confused how that is relevant here.

It was a mistake.

gdifiore, 2025-02-26:
I'm taking a crack at this

delhoume, 2025-03-02:
WifiCredential already stores an array of SSIDs that had a successful connection to.
And looks like they are automatically sorted by latest date.
So instead of giving up  if _lastUsed fails at line 190 wificaptive.cpp, you just have to try next one in the array (loop over _savedWifis)

Maybe 5  to 10 lines of code

gdifiore, 2025-03-02:
@delhoume Yes, that's exactly what I wrote on my local branch, just waiting for my device to come this week before testing and pushing PR.

delhoume, 2025-03-02:
> [@delhoume](https://github.com/delhoume) Yes, that's exactly what I wrote on my local branch, just waiting for my device to come this week before testing and pushing PR.

There might be a bug in saving credentials, the ssid and password are not checked for existence in the array so maybe you will have multiple times the same  pair in _saveWifis and nothing but this pair...

And maybe I would  make readWifiCredentials take an index an index so that it can be looped over.
and I would get  rid of _lastUsedXX, replaced by index 0.

I am not that sure the implementation of the array in WifiCaptive::saveWifiCredentials is correct, it has to be checked ...

gdifiore, 2025-03-06:
Implemented and tested on my end, #76

Omeryl, 2025-03-12:
Fixed with #76.

closed 2025-03-12 by Omeryl

jernaumorat, 2025-06-05:
This does not appear to work on my device running v1.5.5, are there any additional steps needed to enable this?