usetrmnl/trmnl-firmware #78: Add early release setting

bkuhlmann, 2025-03-06:
## Why

Hey. :wave: 

I'm wanting to add the ability to configure OTA support by updating any TRMNL device that makes a `/api/display` request to the [BYOS](https://github.com/usetrmnl/byos_sinatra) server (this could work for Core too).

## How

I'd like to propose including this new key/value in the BYOS `/api/display` response:

``` ruby
early_firmware: true || false
```

This would allow BYOS folks to opt into early firmware releases by using a `true` value (`false` would be the default).

## Notes

If `update_firmware` is set to `true`, this will automatically update the device to the latest firmware, correct? If so, then I'd assume, for this feature to work, you'd have to have both of the following settings enabled:

``` ruby
{
  update_firmware: true,
  early_firmware: true
}
```

ryanckulp, 2025-03-07:
this is supported on Core as `device.accepts_beta_firmware`, where `false` is the default, and configurable from the Device settings page. so i think for now, BYOS should just mimic that implementation.

note in Core > Admin > Firmwares, we have an `environment` state enum:

```rb
enum :environment, [:development, :staging, :production]
```

whenever we tag a release as `staging`, this `*.bin` absolute URL is returned to all devices with `accepts_beta_firmware=true`.

bkuhlmann, 2025-03-07:
Ah, OK, so the URLs are determined by environment. I need to add that to the BYOS DB migration then. Will do.

I'm closing this issue since the work is on my end and not with the Firmware. Thanks!

closed 2025-03-07 by bkuhlmann