usetrmnl/trmnl-firmware #80: Include device ID header with `/api/log` calls

jamesthurley, 2025-03-07:
Currently the access token is included, but not the ID:

https://github.com/usetrmnl/firmware/blob/aa476cbb6db0cff32cc78243e18d347c984a8998/src/api-client/submit_log.cpp#L41-L43

Including the ID would allow BYOS to only show logs from authenticated devices.

Omeryl, 2025-03-08:
Why not just look up the device via the API key it's already providing?

jamesthurley, 2025-03-08:
I mean I could if people think including the ID is a bad idea. I just figured it would likely be a one-liner in the firmware to include the ID header, and then you can authenticate every request in the same way.

jamesthurley, 2025-03-08:
Hmm, well not all devices go through `/api/setup` (see #79), so nuking it from everything else could mean you never see the ID of some devices.

Personally I find it useful in my server, so I'd request at least not removing it from methods that already get it, and adding it to `/api/log` would let me delete some code so that would be nice.

PR #88 (2025-03-11): Add ID header to /api/log call
This adds the ID header to the `/api/log` call.

Reasoning:
- It improves consistency, as both `/api/setup` and `/api/display` currently receive the device ID.
- It allows you to identify a device which has a bad or expired API key from `/api/log` calls.
- It allows looking up a device by ID on all API calls.

Resolves #80

Tested on device.
closed 2025-03-11 by ryanckulp