usetrmnl/trmnl-firmware #81: Firmware calls `/api/display/` rather than `/api/display`

jamesthurley, 2025-03-07:
It also calls `/api/setup/` rather than `/api/setup`.

It correctly calls `/api/log`.

This tripped me up with my BYOS. The very useful TRMNL Simulator calls `/api/setup` and `/api/display` as you would expect, but then I had to add some path normalizing middleware to make it work with the actual TRMNL.

ryanckulp, 2025-03-07:
i'd like to remove all trailing slashes. didn't touch it yet, as [BYOS Sinatra](https://github.com/usetrmnl/byos_sinatra) is picky about URLs and i didn't want to break 1, simplify the other. so this change needs to be done in concert with BYOS endpoints.

bkuhlmann, 2025-03-10:
I'd like this as well. I'm very nearly ready to support `/api/display` instead `/api/display/` and have this working locally for our new Hanami implementation. :tada:

Would it be possible to drop the trailing slashes for a new Beta firmware release so I can do more testing locally and then release both the Hanami app and the new Firmware at the same time?

bkuhlmann, 2025-03-24:
:information_source: This works properly in [Terminus](https://github.com/usetrmnl/byos_hanami) as you can make API requests with or without the trailing slashes. Hopefully this'll make dropping trailing slashes more compelling now.