usetrmnl/trmnl-firmware #83: ETag support

jamesthurley, 2025-03-07:
I think that `usetrmnl.com` will compare the newly generate image with the previously generated image, and if they are the same it won't send a new image to the TRMNL device, saving bandwidth and battery.

In my BYOS I generate screens on the fly and don't cache them anywhere, so I can't compare images, but supporting [ETags](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/ETag) in the firmware would allow servers to implement the same optimization without caching images. It could potentially be a more efficient way for `usetrmnl.com` to do it as well.

The server would send the image hash back as an ETag header along with the image, and the TRMNL would store the ETag and send it back to the API in the headers of the next `/api/display` request. The server can then simply compare ETags to determine if the screen has changed.

Omeryl, 2025-03-08:
I won't lie, this seems kind of overkill when we already can just compare the filename to the current displayed filename (as is what happens, if I recall correctly). This is much cheaper to support on the device itself with regards to power.

jamesthurley, 2025-03-08:
But checking the filename only works if you're generating images in advance.

You have to send the file name at he same time as the `image_url`, so if you're generating the images dynamically, when the `image_url` is fetched, then you can't know if the image has changed or not at the point you send the file name to the TRMNL.

jamesthurley, 2025-03-08:
An alternative would be if the TRMNL supported `data://` URLs for images you could generate them during the `/api/display` call and put the image hash as the filename. That would avoid the need for an ETag, and also avoid the need for a second HTTP request to fetch the image.

Omeryl, 2025-03-11:
Ah, yes, as long as we don't actually compute the ETag it wouldn't be crazy ... but why wouldn't you just do this yourself? Just compute a random image_url yourself, and in your endpoint just use that image name as the "ETag". If it's different than what you expected, dynamically generate it, if it isn't, then don't.

jamesthurley, 2025-03-11:
The ETag only works if the client stores it and sends it back with the next request. Here is an abridged quote from the link in my original request:

> Another typical use of the ETag header is to cache resources that are unchanged. If a user visits a given URL again [...] the client will send the value of its ETag along in an `If-None-Match` header field:

```
If-None-Match: "33a64df551425fcc55e4d42a148795d9f25f89d4"
```

> The server compares the client's ETag (sent with If-None-Match) with the ETag for its current version of the resource, and if both values match (that is, the resource has not changed), the server sends back a `304 Not Modified` status, without a body, which tells the client that the cached version of the response is still good to use (fresh).

To answer your other question:

> Just compute a random image_url yourself, and in your endpoint just use that image name as the "ETag". If it's different than what you expected, dynamically generate it, if it isn't, then don't.

This doesn't work, because to know if the resource has changed or not for the device you need to know what the last ETag the device received was. A random `image_url` doesn't tell you anything.

Omeryl, 2025-03-11:
Sure, but the filename behaves just like an etag for both the device and server. Here's what I mean:

1. First the device hits /api/display, the device then receives a URL for an image. That is arbitrary, and can be set to whatever the BYOS implementation wants. In this case, it can behave _exactly_ like an ETag. If it's the same as it was before, the device will not request a new image.
2. On your BYOS implementation, use something like the following route: /api/generate_screen/{device_id}/{filename}.bmp. The filename can be random, just like an ETag as you want, and if you receive the same URL for something you've already generated (which you shouldn't, assuming your /api/display was consistent in returning the same filename as before) then all is well. I'd recommend storing the last result of your API generation just in case, it's only about 47KB, so that way you can send it back to the device if you got the same request again.

jamesthurley, 2025-03-11:
Ok, I think I see why we're getting our wires crossed, and I realise I've omitted a critical piece of information:

My server is entirely stateless and does no caching. It doesn't have a database or a distributed cache. It doesn't write to the file system. It is nothing but a read-only container containing a binary and some configuration files.

I apologise.. I was reading your responses from within the context of these constraints and wasn't considering that you weren't even aware of them, or that if I didn't have these constraints then there would indeed be other viable options.

As it is, I'm completely dependent on the device telling me what it last received, and using an ETag is the standard way of doing this.

Anyway, I've now implemented the functionality in a PR and it's working nicely on my device. Hopefully it will be considered straightforward enough to be approved.

jgstew, 2025-04-11:
This does make sense to me, etag is the existing standard HTTP way to do this sort of thing, not the filename.

The only issue would be if the web server implements etags incorrectly and returns the same etag for a period of time, even when the asset has changed. Seems like if the webserver doesn't return an etag at all or returns a new etag every time even when it hasn't changed, then the existing filename based logic would work correctly.