usetrmnl/trmnl-firmware #96: Fix API display endpoint blank HTTP_ACCESS_TOKEN header

bkuhlmann, 2025-03-24:
## Why

Hey. :wave: I'm seeing the `HTTP_ACCESS_TOKEN` header being sent to `/api/display` with a blank value (i.e. empty string). It doesn't matter if I'm connected to an existing device or auto-provisioning a new device, the behavior is consistent.

## How

I'm able to reproduce this is to using [Terminus](https://github.com/usetrmnl/byos_hanami) Here's the response after the device hits the `/api/setup` endpoint:

``` 
{
  "api_key": "<redacted>",
  "friendly_id": "A62C92",
  "image_url": "http://192.168.0.74:2300/assets/setup.bmp",
  "message": "Welcome to TRMNL BYOS."
}
```

I've removed the API key provided by the server in the above example but for some reason this isn't being recorded by the device when it makes API requests to `/api/display`. Is the device somehow not recording the `api_key` value when it makes subsequent `/api/display` requests?

## Notes

* **Firmware**: 1.4.7

Omeryl, 2025-03-24:
Do you have a minimum set of code for reproduction?

FractalMachinist, 2025-04-20:
I also see this issue.

# Summary
1. TRMNL devices do not provide their API key when requesting `/api/display`.
2. TRMNL devices might not keep their API key at all: they call `/api/setup` before `/api/display` as though they never finish setting up (perhaps)
    1. Perhaps [this](https://github.com/usetrmnl/firmware/blob/2831ea85216f3873009022fcf11c59c51e26c083/src/bl.cpp#L492) provides a default which is being used too often?
    2. Perhaps [this](https://github.com/usetrmnl/firmware/blob/2831ea85216f3873009022fcf11c59c51e26c083/src/bl.cpp#L307) is related?
    3. I don't see an easy way for me to get the TRMNL device internal logs via USB, otherwise I'd look there.
3. TRMNL devices ignore newer images on the same filename if that is currently the only filename in that device's rotation. Example:
    1. TRMNL device boots; Server boots; `clock.bmp` has been added to the server.
    1. TRMNL device hits `/api/display` --> `{"image_url": "https://.../clock.bmp", ...}`
    2. TRMNL device downloads and displays `https://.../clock.bmp`
    3. TRMNL device sleeps
    4. Periodically, `/api/screens` receives a submission and replaces `clock.bmp`.
    5. TRMNL device hits `/api/display` --> `{"image_url": "https://.../clock.bmp", ...}`
    6. TRMNL device "Oh, I'm already displaying that, let's save battery"
    7. TRMNL device sleeps
    8. (Process repeats)
    9. TRMNL device forever shows the first `clock.bmp` it came across because it doesn't see when a file was last modified.

# Context
- **Firmware**:
  - 1.4.7
  - 1.4.8
  - 1.5.1
    - [firmware/builds/FW1.5.1.bin](https://github.com/usetrmnl/firmware/raw/2831ea85216f3873009022fcf11c59c51e26c083/builds/FW1.5.1.bin) at 2831ea85216f3873009022fcf11c59c51e26c083
    - I downloaded the built binary and installed it OTA from the Terminus server
    - The logs below were generated on 1.5.1
- **Server:** [Terminus 0.6.0](https://github.com/usetrmnl/byos_hanami/releases/tag/0.6.0)

# Thorough Process
0. (At any of the above firmware, and perhaps others)
1. Set up the Terminus server on a local network.
    1. The Postgres DB can be empty at this point, or it can already have been used by Terminus; it makes no difference.
    2. Sometimes the Terminus server errors out if the postgres db's `/var/lib/postgresql/data` starts off empty, but a restart always clears that up.
3. When the TRMNL device is powered on, long-press the button to reset it.
    1. _Resetting the device is not required. I'm just being reproducible._
5. Connect to the TRMNL device's wifi and configure it with the captive portal:    1. Optionally: tell the device to perform a soft reset, then re-connect. This makes no difference.
    2. Input valid WiFi network and credentials
    3. Configure the custom server above
    4. Submit
4. Observe: the TRMNL device screen changes and displays the TRMNL logo and TRMNL text (expected)
    1. We don't have any screens for it yet
5. Observe: the TRMNL device keeps alternating between hitting `/api/setup` and `/api/display`, but **hits `/api/display` without providing an API key**. Evidence for no API key:
    1. Observe `WHERE (\"devices\".\"api_key\" ILIKE '%%')` in the postgres query which completes a few milliseconds before the `/api/display` request completes.
    2. I did not have a chance to look at network traffic to determine whether the TRMNL device provides a blank API key or no API key at all.
6. Submit one test screen.

```sh
curl -X "POST" "http://<redacted>/api/screens" \
    -H 'Access-Token: <redacted API key from the logs and the HTML site>' \
    -H 'Accept: application/json' \
    -H 'Content-Type: application/json' \
    -d '{"image": {"content": "<p>Test</p>", "file_name": "test"}}'
```
7. Observe: The TRMNL visits `/api/setup` --> `/api/display` --> `/assets/screens/{MAC}/{file_name}.bmp`, and displays the test image. (Intended behavior)
8. Adjust the TRMNL's refresh rate to 10 seconds. Short-press the button to manually refresh.
9. Observe: The TRMNL continues to visit `/api/setup` before `/api/display` every time it tries to refresh. (Not intended behavior)
10. Submit a new screen at the same filename.

```sh
curl -X "POST" "http://<same>/api/screens" \
    -H 'Access-Token: <same>' \
    -H 'Accept: application/json' \
    -H 'Content-Type: application/json' \
    -d '{"image": {"content": "<p>Test 2</p>", "file_name": "test"}}'
```

11. Observe: The TRMNL device continues to hit `/api/setup` before `/api/display`, but never downloads the newer image for the screen.
12. Short-press the button or power-cycle the TRMNL device.
[... 123 lines cut ...]
2025-04-20T13:17:01.469079614-04:00 web             | {"progname":"terminus","severity":"INFO","time":"2025-04-20T17:17:01Z","message":"  Loaded :postgres in 0ms SELECT \"devices\".\"id\", \"devices\".\"friendly_id\", \"devices\".\"label\", \"devices\".\"mac_address\", \"devices\".\"api_key\", \"devices\".\"firmware_version\", \"devices\".\"firmware_beta\", \"devices\".\"wifi\", \"devices\".\"battery\", \"devices\".\"refresh_rate\", \"devices\".\"image_timeout\", \"devices\".\"width\", \"devices\".\"height\", \"devices\".\"setup_at\", \"devices\".\"created_at\", \"devices\".\"updated_at\", \"devices\".\"proxy\", \"devices\".\"firmware_update\" FROM \"devices\" WHERE (\"devices\".\"mac_address\" ILIKE '%CC:BA:97:60:8D:8C%') ORDER BY \"devices\".\"id\""}
[... cut ...] esh_rate\", \"devices\".\"image_timeout\", \"devices\".\"width\", \"devices\".\"height\", \"devices\".\"setup_at\", \"devices\".\"created_at\", \"devices\".\"updated_at\", \"devices\".\"proxy\", \"devices\".\"firmware_update\""}
2025-04-20T13:26:32.679541823-04:00 web             | {"progname":"terminus","severity":"INFO","time":"2025-04-20T17:26:32Z","verb":"GET","status":200,"ip":"192.168.8.231","path":"/api/display/","length":"-","params":{},"elapsed":33390,"elapsed_unit":"µs"}
```
</details>