usetrmnl/trmnl-firmware #97: Add emulation

bkuhlmann, 2025-03-24:
## Why

Hey. :wave: I don't know if this is a difficult request or not but would it be possible to add emulation support so we could run the firmware via an emulator? The reason I ask is because this would allow me to build more of an integration (end-to-end) test suite. This would not only be handy for local development but also for use in a CI environment. The main target for this request is BYOS but could be quite handy for Core as well.

## How

I think, at a minimum, there'd need to be API support for the emulator to do the following programmatically within a test environment:

- Single, double, and long (5 second) button press/click.
- On/off toggle.
- Multi-version emulation support (or at least being able to pick which firmware version you want to emulate).
- Captive portal emulation but only from the standpoint of supplying WiFi credentials, doing a soft reset, and supplying a custom server URL (especially important for BYOS).

jrr, 2025-04-12:
I'd be a lot more confident making changes to the firmware if there were integration tests for API interaction!

This looks promising: https://wokwi.com/

bkuhlmann, 2025-04-12:
Yeah, no kidding. Happy to figure out how to make this work with [Terminus](https://github.com/usetrmnl/byos_hanami) as a start. Wokwi definitely looks interesting and the pricing id good too. Looks like most CI examples are with GitHub so not sure how difficult this would be to wire up with Circle CI. ...but, yeah, this would be nice to have.