usetrmnl/trmnl-firmware #107: Feature request: Refresh screen immediately if power connected

jgstew, 2025-04-11:
would be interesting and useful if the moment power was connected to charge the trmnl if it would refresh what is displayed.

This would be useful to be able to trigger easily when making changes on what should be displayed so you don't have to wait for the next polling interval of the device.

Omeryl, 2025-04-12:
I am not sure I understand, that's what pushing the button one time on the back does, do you mean something else?

jrr, 2025-04-14:
I interpret this to mean, that the device would wake when power is connected. (in addition to waking on timer and waking on button push)

I'm not sure if this is possible. https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html