usetrmnl/trmnl-firmware #109: Need Guidance: How to introduce esp32dev environment to the codebase

matrushka, 2025-04-11:
I've been playing with the firmware to run it on a [waveshare esp32 e-paper driver board](https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board). I could finally get it working and would like to create a PR with my changes but I'm a bit lost. Seems like majority of the changes I needed was in DEV_Config.cpp, the one in the main branch is not working at all (reboot right after `src/display.cpp [18]: dev module start`) so needed to replace it with the original one from [here](https://files.waveshare.com/upload/5/50/E-Paper_ESP32_Driver_Board_Code.7z). I know how to get the smaller changes handled with a build flag but I don't really know how to handle DEV_Config.cpp part (which seems to start differing from the original one at [this release](https://github.com/usetrmnl/firmware/commit/f67f667f0a3061c260e6b5ffa65fd7775e410050#diff-751243070aecfa2dcdffb7bd5664b38f2b26b4e9cc70d3ba769775873131e560)). I'd be happy to continue my work and create a PR with some guidance.

Phaestion, 2025-04-19:
Hi @matrushka! I'm trying to get this running too. Could you upload your fork? I got the build working but it does not seem to display anything. If you already did the ground work on this and got it working, it would save me a lot of time to look at this.

I'd also be happy to help make the PR, while I'm mostly a Java/Kotlin dev I used to be a C++ dev in my previous life so I might be able to help. I guess this would also depend on why the TRMNL devs actually changed the DEV_Config.cpp of Waveshare.

matrushka, 2025-04-20:
Hey @Phaestion here's my branch, jsut rebased in latest commit. Didn't have the chance to try it but I assume it should work fine (Don't forget to use the `esp32dev` environment in platformIO): https://github.com/matrushka/usetmnl-firmware/tree/esp32. It has few caveats though:
* As the board does not have buttons, you cannot reset the device easily. I've tweaked the API server to return reset the  firmare to do it.
* Had to disable `esp_deep_sleep_enable_gpio_wakeup`
* I could not test any battery related functionality

When I checked the release log for the changes to DEV_Config.cpp, I've noticed this line: `Made several optimizations to make power cycle last faster (~2.5 seconds)` so I assume that was the reason they changed it so the screen displays something faster (although this is just a very rough guess).  Let me know what do you think and if there's somerthing I can help with.

Phaestion, 2025-04-22:
Thanks @matrushka! I could get it working on my own just looking at your first comment, it was a bit of a learning curve getting up to speed with the platformIO configurations. I had a look at your fork and it looks better than what I hacked together, I'll see what I can do in the next few days to get this ready for a PR.

matrushka, 2025-06-09:
https://github.com/usetrmnl/firmware/commit/481d11626171a8153ca2088dec8a4dbec9fc8ddb solves this, just tried 1.5.5 and works fine