usetrmnl/trmnl-firmware #112: firmware BMP handling vs PIL

eichin, 2025-04-13:
(firmware 1.4.8, device arrived this week; PIL is `python3-pil` 10.2.0-1ubuntu1.)

The PIL `BmpImagePlugin.py` `save()` function does
```python
   if im.mode == "1":
        palette = b"".join(o8(i) * 4 for i in (0, 255))
```
The `lib/trmnl/src/bmp.cpp`  in this repo, `parseBMPHeader` function only succeeds if one of
```C
    if (data[54] == 0 && data[55] == 0 && data[56] == 0 && data[57] == 0 && data[58] == 255 && data[59] == 255 && data[60] == 255 && data[61] == 0)
...
    else if (data[54] == 255 && data[55] == 255 && data[56] == 255 && data[57] == 0 && data[58] == 0 && data[59] == 0 && data[60] == 0 && data[61] == 0)
```
are true, otherwise `BMP_COLOR_SCHEME_FAILED` is returned/logged.

So, the firmware is looking for a white entry that is [0,0,0,0] and a black entry that is [255,255,255,0] but PIL  is generating [255,255,255,255] for black.  I'm not asserting that one or the other is wrong, just starting here since the firmware is the one complaining about the discrepancy.

The workaround from the python side is to simply clear that byte before sending the BMP across:
```python
    img.save(outbuf, "BMP")
    retval = bytearray(outbuf.getvalue())
    # firmware wants 0*4, 255*3+0, but BmpImagePlugin.py uses 0*4, 255*4
    retval[61] = 0
```
[Wikipedia](https://en.wikipedia.org/wiki/BMP_file_format#Color_table) suggests that that fourth byte Must Be Zero - perhaps in this case, checking only the R/G/B entries would be reasonable in the name of "generous interoperability".  (A quick look at the [PIL upstream github issues](https://github.com/python-pillow/Pillow/issues?q=is%3Aissue%20BmpImagePlugin) doesn't turn up any open or closed tickets related to this so I will head there next - but there's a much longer lag between PIL changing and casual developers actually getting the updated code, than firmware changes.)

Thanks!

eichin, 2025-04-13:
For completeness, https://github.com/python-pillow/Pillow/issues/8888 is the corresponding PIL issue discussing whether that fourth byte really should be zero and whether `BmpImagePlugin._save` needs to change.

jrr, 2025-04-13:
The actual bitmap rendering logic doesn't even look at the header:

https://github.com/usetrmnl/firmware/blob/76813028651efd05e24791109b87a23da44b0bfc/src/display.cpp#L103

(`Paint_DrawBitMap` is Waveshare library code)

So we ought to be able to ignore that last byte as you suggest. Would you be up for making a contribution? This happens to be an area of the app covered by unit tests, so you could start with a new case in `bmp.test.cpp`.

delhoume, 2025-04-13:
@jrr  yes, all that matters is the bit value that maps to black or white.

if you want an inverted image (Black and White inverted) you invert the bits in the data and  do not change anything to the  (unsed) palette

palette entry does not matter (but it will when there will be supported devices with more than 2 colors.

note that 1 bit BMP does not mean N&B but 2 bgra colors.

delhoume, 2025-04-28:
My https://github.com/usetrmnl/firmware/pull/125 PR addresses this (and a few other BMP related issues)

closed 2025-08-31 by ryanckulp