usetrmnl/trmnl-firmware #131: What is the compatible BMP format that the trmnl firmware accepts?

thomaswmb, 2025-05-11:
The firmware seems to be very picky regarding the BMP files provided.
I cant produce a correct BMP from photoshop to be accepted by the firmware.
I am following the spec with 800x480 1 bit.
There are just 3 different options in PS when saving to that format (Windows / Windows with flipped row order / OS2) and none of them seems to save a correct file (at least according to the firmware)
Comparing the test.bmp from this repo and my PS files, there is a difference of 2 bytes.
Looking into the files side by side in a text editor shows that they seem to be very different.

The server implementations seem to use image magick to create the BMP files for the device.
Is this the only supported way to generate compatible images?

PS: It took me a long time to figure out that the BMP file i provided was the real issue with getting anything displayed. The log requests from for the bad BMP files over http state just this:

`    <log_message>HTTPS request error. Returned code - 200, available bytes - 48064, received bytes - 0</log_message>`

Only after i started looking at the code in bl.cpp i have discovered the internal log message for this error that states: 

`"Receiving failed. Bad file size".` 

This made me dig and compare the files i created VS the test.bmp. Think the log message could be improved.

ryanckulp, 2025-05-11:
check this out and let me know if we need to make improvements, can also add to our README here:

https://docs.usetrmnl.com/go/imagemagick-guide

thomaswmb, 2025-05-11:
hey thanks for the fast response. I should have read the docs more caerfully. i knew i have seen something about IM on my way.
anyway it would be nice if the firmware could handle different types of BMPs, not just BMP3 if possible.
btw i had also no luck with PNGs from photoshop :(
since BMP3 is the "native" windows format and i plan to control my termnl device from a .NET webserver i will try to craft a proper BMP with .NET on the fly. just want to avoid image magick alltogether. wish me luck;)

delhoume, 2025-05-11:
for some reason (maybe padding), PS adds 2 bytes to the file, and the field biSizeImage is set accordingly.

So the  image is valid, because the spec does not prevent additional data to be there after the image data.

TRMN firmware enforces overall size of 48062, so this image  is rejected. It could be accepted with small changes to the source code.

You can try to remove the 2 last bytes and patch the two references to 40002 and 48064.
Some software do not really care about biSizeImage as the BMP can be decoded perfectly with 0 as values (and for uncompressed bmps suchas 1 bit ones the spec even says it should  be 0)

[image]

[image]

[image]

delhoume, 2025-05-11:
I have submitted code for FW that can handle BMP3 BMP4 and BMP5, is less picky about palette alpha, handles negative height, etc...

https://github.com/usetrmnl/firmware/pull/125

thomaswmb, 2025-05-14:
How about also accepting just the raw image data without the bmp header? i mean as an optional way to provide these 1 bit images? That would be pretty neat for server setups that generate images in memory only, not handling files or converting with image magick. I am doing just that right now (reading raw bytes back from GPU where my image lives as texture), but of course need to add the 62 bytes of header in front of the data before it can be served on the endpoint. this is no big deal of course and probably a pretty unusual setup but handling the header feels kind of superfluous in my case. Thanks for considering.

delhoume, 2025-05-14:
The header might seem superflous but it is highly unusual to transmit raw data without a mime type. 
if you do no have a header you have to know the format (width, height, bitdepth) or you will not be able to distinguish between 10x100 and 100x10.

I was not aware the GPU can use 1 bit as a texture, and I am surprised the raw data is in 1 bit BMP format on the GPU, because of the BMP padding to 32 bits, while tendency is to pad to nearest byte and because smallest value usually a byte.
And current tend is to have at least 32 bits RGBA,GPU dedicated ram has tremendously grown the last years (mostly because of the IA crazyness).
I did some OpenGL in the past (that would be around 1997 with OpenGL1.0 and 1.1 ;-)

Anyways, generating a 1 bit  BMP header is very easy in C/C++, you just have to declare 3 or 4 struct, fill the with default values, and only poke into at most 5 or 6 fields. Or you can even have a 
static char header[62] = { 'B', 'M', ...}; and patch it 
The wikipedia page for Windows BMP is very well documenting the format.
 
and poke into it (beware of endianness fr some machines).
 
My advice is to handle actually more BMPs and PNGs types payload, with a size limit in bytes (no more than the largest 1 bit bmp - that would be about 48146, save this to flash, then reload it to a framebuffer that is is in the native format for the panel.
You have 2 buffers but you can receive and display independently, convert if needed  (you could receive a 4 bit greyscale png and convert it to 2 bit if your panel only handles 4 grey levels and so on (and no need to fork the drivers code).

I have a fork with code that handles BMP just like PNGs with the structs, I don'nt think it can be used as is but it gives an idea of what can be done.
https://github.com/delhoume/firmware/blob/noflip/src/bmp.cpp

The noflip branch in my fork has lots of modified code to  be able to implement my ideas
https://github.com/usetrmnl/firmware/compare/main...delhoume:firmware:noflip

The main issue with current  firmware implementation is that the same buffer is used to receive the image and to display it, assuming it will always be in a format directly compatible with the screen.

it is in fact a little more complicated than that, because the current firmware is modified to comply with a specific 1 bit bmp payload that is not the exact spec (some valid BMPs are no accepted because of this, this was one of the motivations for this PR)

thomaswmb, 2025-05-15:
you are right, no meta data is bad. when i think about it again, just RAW image data is not the best idea regarding trmnl beeing open to different screens an such. nevermind. it would just work for the current official model but not for any other devices or later models.
and yes there is no thing like 1 bit textures possible on gpus. best you can do is one channel but still 8bit. i need to loop over the bytes and threshold to 1bit after readback. then just adding the header part of test.bmp, i am lazy:)
thanks for your effort to widen the image compatibility!

ryanckulp, 2025-08-31:
closing because we no longer use BMP

closed 2025-08-31 by ryanckulp