usetrmnl/trmnl-firmware #132: Fix setup to display API request bounce

bkuhlmann, 2025-05-12:
## Why

When a device is connected to a BYOS system, like [Terminus](https://github.com/usetrmnl/byos_hanami), I constantly see the following API requests in the logs:

- GET 200 /api/setup
- POST 204 /api/log
- GET 200 /api/display

The problem is that my device logs constantly fill up with these:

<details>
  <summary>Log Entries</summary>

[image]
</details>

<details>
  <summary>A Specific Log Entry (example)</summary>

[image]
</details>

I'm not seeing anything else in the logs to indicate why a device thinks it can't resolve the host name. I constantly get that log entry, yet, my device continues to serve up images without issue.

Something seems broken between the Setup API response and the Device API request. When the device is already setup, why does it continue to make Setup API requests?

## How

The easiest way to recreate this would be to run Terminus via Docker and connect your device to it and and then set the device refresh rate to something fast like 10 or 50 seconds. The steps would be roughly:

- `git clone https://github.com/usetrmnl/byos_hanami terminus`
- `cd terminus`
- `docker-compose up`
- Then follow the [device provisioning](https://github.com/usetrmnl/byos_hanami?tab=readme-ov-file#device-provisioning) steps.

## Notes

- **Firmware**: 1.5.4.
- This is related to [Issue 96](https://github.com/usetrmnl/firmware/issues/96) which provides even more detail.

bkuhlmann, 2025-05-13:
:information_source: I've updated the above to specify that these issues started with Version 1.5.3 after forcing my device to update to the older versions to detect where this issue first cropped up. I discovered that **Version 1.5.2** is the latest stable firmware release where these issues *don't appear*.

Omeryl, 2025-05-13:
This is a preflight check I introduced to resolve the hostname before trying to connect which helped out a lot of users. For users with an IP as their API url it would break, which is what it sounds like is happening here. PR is welcome to fix but it's purely informative and you could probably drop the logs on your end in the meantime.

bkuhlmann, 2025-05-13:
Thanks. Yeah, this is becoming a major issue as I'm getting an uptick in Terminus issues related to this change. I've added a [warning](https://github.com/usetrmnl/byos_hanami?tab=readme-ov-file#terminus) to the top of the Terminus documentation to let people know they should stick with Version 1.5.2 until this is fixed and I've added [additional documentation](https://github.com/usetrmnl/byos_hanami?tab=readme-ov-file#firmware) explaining how people can use Terminus to revert to 1.5.2 if they are using the newest versions.

We need to get this fixed because these `/api/log` requests are filling up the device logs for folks on nearly every `/api/display` request. I haven't written any C code since college so wouldn't be qualified to fix this myself otherwise I would attempt to tackle this.

I also need to capture these logs for each device because they are helpful in debugging why a device can't connect to Terminus. Without these logs, I has less insight as to what's working (or breaking) with each new firmware release.

Omeryl, 2025-05-13:
@bkuhlmann PR welcome to fix this :)