usetrmnl/trmnl-firmware #133: Add log levels

bkuhlmann, 2025-05-14:
## Why

Due seeing an increase in log levels since the release of Version 1.5.3, I'd like to suggest adding log levels to all `/api/log` requests (this also came up in team discussion during standup). Typically, this would be the following (starting with the most verbose to least verbose):

- **Debug**: Includes all log levels and is the most verbose.
- **Info**: Includes informational messages.
- **Warn**: Includes warning messages. Definitely worth being concerned about but should not break anything.
- **Error**: Includes error messages. These are bad not not catastrophic.
- **Fatal**: Includes fatal messages. These are catastrophic.

## How

Currently the log payloads look as follows (although, it would be nice to clean this up further as documented in this [issue](https://github.com/usetrmnl/firmware/issues/99)).

<details>
  <summary>Current Payload Example</summary>

ℹ️ The following uses Ruby hash syntax only for demonstration purposes.

``` ruby
{
  log: {
    logs_array: [
      {
        device_status_stamp: {
          battery_voltage: 4.772,
          current_fw_version: "1.2.3",
          free_heap_size: 160656,
          max_alloc_size: 990000,
          refresh_rate: 30,
          special_function: "none",
          time_since_last_sleep_start: 31,
          wakeup_reason: "timer",
          wifi_rssi_level: -54,
          wifi_status: "connected"
        },
        additional_info: {
          retry_attempt: 1
        },
        log_codeline: 597,
        log_id: 2,
        log_message: "Danger!",
        log_sourcefile: "src/bl.cpp",
        creation_timestamp: 1742000523
      }
    ]
  }
}
```
</details>

Given the above, this means we'd need to add a `log_level` key. Example (truncated for brevity and using Ruby syntax for demonstration purposes):

``` ruby
{
  log: {
    logs_array: [
        log_level: "info"
      }
    ]
  }
}
```

We'd need a way to update the device to provide different log levels as well so Core, and especially BYOS, could let the device be either more verbose or less. For BYOS, this would be powerful because we could start at log level INFO and when customers hit issues they could set log level to DEBUG to help file better bug reports.

Initially, this could be done by sending the desire log level as part of the API Display response to the device. Example (truncated for brevity):

``` json
{
  "log_level": "debug"
}
```

## Notes

This change would be of great help when debugging issues with customers both via Core but especially for BYOS.

jrr, 2025-05-14:
Today there are two kinds of logging:
```cpp
// send to the server else persist for later
submit_log("HTTP Client failed with error: %s", https.errorToString(httpCode).c_str());

// print locally
Log_info("Firmware version %d.%d.%d", FW_MAJOR_VERSION, FW_MINOR_VERSION, FW_PATCH_VERSION);
Log_error("JSON deserialization error.");
Log_fatal("preferences clearing error");
```

Also there's still some use of the underlying ArduinoLog, though [I'd like to clean those up](https://github.com/usetrmnl/firmware/pull/121):

```cpp
Log.info("Trying to autoconnect to wifi...\r\n");
```

It'd be nice to combine these two concepts into a single one-logger-to-rule-them-all. 

One challenge is that sending a log over the network, _while you're already in the middle of a network interaction_, can consume too much stack. So it may make sense to separate the "persist log locally" and "flush logs to server" steps.

jrr, 2025-05-14:
## Proposal

1) We keep the current signatures of `Log_info` / `Log_error` / `Log_fatal`. (and perhaps add "debug" and "warn" as @bkuhlmann suggested)
2) There's a new cloud-managed `log_level` setting that governs what severity gets stored / reported. When you call one of the logging methods, based on the setting it will either a) just log it locally over serial, or b) log it and also also [store it](https://github.com/usetrmnl/firmware/blob/3366fb1d201241bbc99c43e475e588982dd9edd2/include/stored_logs.h) for later submission. 
3) We [flush the stored logs to the server](https://github.com/usetrmnl/firmware/blob/1344b71f2c9c3ffe6fbbe3e23f52289a2283b93b/src/bl.cpp#L1868) more often, between other network requests.

This would avoid the issue we have today where the app can have two pairs of WiFiClient / HTTPClient alive at once, which I think has led to problems.

One concern with this approach (storing to flash more aggressively) is that it may burn out the flash faster? It looks like ESP32's [NVS has wear-leveling](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html) so maybe it's not a big deal.

ryanckulp, 2025-05-20:
thanks JRR for explaining and for the ideas. to MVP this we would benefit tremendously on the server side (BYOS too) with a new `log_level` attribute in the payload body, which can mirror the `Log_<type>` already in FW. 

this will power user-facing sort tools and let us safely ignore noise. currently if a user sees a Log, they assume it's an error.

jrr, 2025-07-20:
To square the levels identified above with [ArduinoLog's](https://github.com/thijse/Arduino-Log/blob/3f4fcf5a345c1d542e56b88c0ffcb2bd2a5b0bd0/ArduinoLog.h#L40-L47), here's a table:

| ArduinoLog level  | Currently used by trmnl-firmware | Description |
| ------------- | ------------- | --|
| VERBOSE  | ✅  | Debug: Includes all log levels and is the most verbose. |
| TRACE  | 🚫  |
| NOTICE  | 🚫 |
| INFO  | ✅  | Includes informational messages. |
| WARNING  | 🚫  | Includes warning messages. Definitely worth being concerned about but should not break anything. |
| ERROR  | ✅  | Includes error messages. These are bad not not catastrophic. |
| FATAL | ✅  | Includes fatal messages. These are catastrophic. |

So, to align the approaches, we could:

- add support for _Warning_
- "alias" _Verbose_ to _Debug_. We can call it 'debug' in our code, but it'll still appear as 'V' in [ArduinoLog's output](https://github.com/thijse/Arduino-Log/blob/3f4fcf5a345c1d542e56b88c0ffcb2bd2a5b0bd0/ArduinoLog.h#L357).

bkuhlmann, 2025-07-20:
Thanks, this is helpful. Yeah, I'd vote for adding *warn* and aliasing *verbose* as *debug* for the value of the `level` key (*I've also updated the original issue description to reflect the current log payload data structure as well*). Assuming you're OK with this, that'd mean the `level` key could be any of the following values:

- debug 
- info
- warn
- error
- fatal

ℹ️ At some point it might be nice to have a *trace* level for advanced debugging purposes but we can hold off, for now, and see how far we get with the *debug* log level.