usetrmnl/trmnl-firmware #156: Rendering images from endpoints without content-length response header

ryanckulp, 2025-06-06:
there are 2x plugins that support pointing a device to remote image content, e.g. `personal-website.com/chart.bmp` or `192.168.1.1/image.png`.

https://help.usetrmnl.com/en/articles/10701448-alias-plugin
https://help.usetrmnl.com/en/articles/11035846-redirect-plugin

a user claims their TRMNL could not render their image (just showed logo + skipped to next Playlist item) until the `image_url` response include a `content-length` header.

**steps to reproduce**
1. generate an 800x480 PNG that [conforms to these standards](https://docs.usetrmnl.com/go/imagemagick-guide)
2. host it somewhere without (empty?) `content-length` response header
3. [create an Alias plugin instance](https://usetrmnl.com/plugin_settings/new?keyname=alias) with the image's URL
4. drag this plugin instance to be next up in your Playlist
5. click btn on back of TRMNL to skip to this item

gabe565, 2025-06-06:
I'm the person that reported this.

The reason it happened is because I wrote my server in Go, and I was streaming the image encode output directly to the client. This meant Go wasn't aware of the file size until it was done encoding, and at that point it was too late. In my case, I made it buffer the encode and it fixed my issue.

The firmware's HTTP library may require the header so it can allocate an appropriately-sized buffer. If it's possible to fix then it would improve server compatibility, but I'd say it's also not a deal-breaker if you can't.

foxxyz, 2025-06-20:
That's good to know! I'm using the device in a context where it will never see the internet and have a custom web server answering queries inside a private LAN, so I was just reading the [firmware docs](https://github.com/usetrmnl/firmware?tab=readme-ov-file#web-server-endpoints) for information on how to mimic all the expected endpoint responses.

What was surprising was that it wasn't even posting anything to /api/logs, it just kept trying the image url without an indication about what was wrong.