usetrmnl/trmnl-firmware #173: Voltage -> SoC conversion

simoncozens, 2025-06-29:
Has anyone determined a formula for battery state of charge? It would be nice to be able to show something more useful than voltage in the server logs.

ryanckulp, 2025-06-29:
here's what we use on the web app. it's not perfect. after a few refresh cycles, full voltage (~4.1) will quickly be reflected as 95, 90, 85%.
 
```rb
  def percent_charged
    self.battery_voltage ||= 100
    pct_charged = ((battery_voltage - 3) / 0.012).round(2)

    case pct_charged
    when 88..1000000 # 4.08v follows a full charge
      100.0
    when 85..88
      95.0
    when 83..85
      90.0
    when 10..83
      pct_charged.round(2)
    when -1000000..10
      1.0
    end
  end
```

simoncozens, 2025-06-30:
Super helpful, thank you!