usetrmnl/trmnl-firmware #191: pool.ntp.org hardcoded which goes against ntp.org policy

julemand101, 2025-07-17:
I can see in `bl.ccp` that the timeserver pool `pool.ntp.org` are being hard-coded:

https://github.com/usetrmnl/trmnl-firmware/blob/c13aa80d9dd2ff3ffc0d6f9723e31a3e8e625429/src/bl.cpp#L1863-L1869

This goes against the ntp.org's policy for vendors: https://www.ntppool.org/en/vendors.html

Which have following sections:

>  You must get approval from the server operator before you hardcode any IP addresses or hostnames. This is easy to get if your own organization runs the NTP servers you are planning to use. In most other cases you will not get it.
>
> Do not use the standard pool.ntp.org names as a default configuration in your system. The NTP Pool can offer services for you, but it must be setup in advance (see below).

And:
> **Get your vendor zone**
>
> To allow you to use the pool as the default time service in your application, we will set you up with special hostnames, for example 0.vendor.pool.ntp.org, 1.vendor.pool.ntp.org, 2.vendor.pool.ntp.org and 3.vendor.pool.ntp.org.
>
> You **must absolutely not use the default pool.ntp.org zone names** as the default configuration in your application or appliance.
>
> You can[ apply for a vendor zone](https://manage.ntppool.org/manage/vendor) here on the site.
>
> If you have an open source ntpd implementation or an operating system including ntpd, we ask that you make a reference in the configuration file or documentation encouraging people to [join the pool](https://www.ntppool.org/join.html).
>
> If you are a commercial / closed-source vendor we will ask for a [small contribution](https://www.ntppool.org/vendors/contribute.html) to help scaling the pool to meet demands.
>
> Questions? Email [ask@develooper.com](mailto:ask@develooper.com).

And:
> **Open source projects**
>
> Open Source projects are of course particularly welcome to use the pool in their default setup, but we ask that you [get a vendor zone](https://www.ntppool.org/en/vendors.html#vendor-zone) when using the pool as a default configuration.

stevekennedyuk, 2025-07-17:
Also ntp servers should be honoured in the DHCP options if it's there

Steve

(I run a stratum 1 ntp server at home and would prefer clients to connect
to that locally)

julemand101, 2025-07-17:
Best to create a separate issue about that to not mix up things. :)

Omeryl, 2025-09-10:
> The two time servers you now use (Google and Cloudflare) have different behavior for handling leap seconds.

Yup, that's definitely potentially a problem in, say, a server context but in this case that leap second shouldn't be detrimental (logs that have no sorting based on timestamp externally).