usetrmnl/trmnl-firmware #203: OTA Updates "Brick" XIAO 7.5" ePaper Panel

mmcnama4, 2025-07-30:
Ran into this issue w/ one of my displays. 

**FW** 
v1.5.6

**Description**
If auto-firmware updates are turned on for a new device, specifically a XIAO 7.5" ePaper Panel, the device will successfully download and possibly install the new firmware, but it does not successfully reboot. It gets stuck on the "will reboot soon" screen.  

**Steps to Reproduce**
There are two ways this can manifest. The first is if you have a BYOD device already added to your account and you turn on OTA Updates. 
1. Add a new BYOD to your account
2. Turn on the `OTA Updates Enabled` option on that device's settings page
3. Load v1.5.6 to the XIAO 7.5" ePaper Panel via https://usetrmnl.com/flash
4. Setup device w/ network credentials
5. Wait for device to automatically reboot/connect
6. Device says `FW update available! Starting now...` presumably pulls the file 
7. Device says `Firmware Update success. Device will restart..`
8. Device never reboots, no button and/or switch combination will get it to reboot

The second way this happens is if you turn on OTA updates for an existing device and reboot. 
1. Add a new BYOD to your account
3. Load v1.5.6 to the XIAO 7.5" ePaper Panel via https://usetrmnl.com/flash
4. Setup device w/ network credentials
5. Wait for device to automatically reboot/connect
6. Device inherits the screen from the web app. 
7. Toggle on 'OTA Updates Enabled'
8. Restart device
9. Device says `FW update available! Starting now...` presumably pulls the file 
10. Device says `Firmware Update success. Device will restart..`
11. Device never reboots, no button and/or switch combination will get it to reboot

*Workaround**
To unbrick the device:
1. Go to the device in the web app
2. Toggle off `OTA Updates Enabled'
3. Hit `SAVE`
4. Plug into USB
5. Go to: https://usetrmnl.com/flash
6. Follow steps to flash, you may need to hit the reset button on the device to get it to appear on the device list
7. Reconfigure wifi credentials. 
8. Device should reboot into your selected screens

mmcnama4, 2025-07-31:
@ryanckulp - Admittedly, I didn't read the docs well enough, and ultimately, I hear your point (RTFM). My take is that a robust UX, unfortunately, must assume the user doesn't read the docs... because they don't. 

The default settings do turn OTA updates off, but then there are no follow-up indicators in the app warning the user that this may not be a good option in some scenarios. 

If this were my app, I'd think about the following to improve the UX:

- On the BYOD code page, it's a good opportunity to share information like not all devices support OTA updates and any other info that might be pertinent to BYOD.
- Better information on the settings page in the OTA section. A "NOTE:" clarifying what devices may/may not be supported would be great (link to a wiki/doc if the list ever grew too far). 
- In-app linking to relevant context in documentation and better tagging/SEO of those docs (when I search BYOD or OTA in the help docs, I can't find anything about this firmware behavior. Similarly, I couldn't find anything clear about this in the developer docs. I'm sure it is somewhere but even after looking for it, I couldn't find anything. It is, however, very clear in the Seeed Studio wiki for this product. 
- A warning that pops up when you try to toggle on or save the toggled-on state for a BYOD device, indicating it may not be supported. Better yet, if the app knows the device type and knows it isn't supported, be more explicit. 
- https://usetrmnl.com/flash is another opportunity to communicate limitations of your own firmware; it's unclear if the TMRNL-provided firmware is considered example firmware or not, and/or why OTA isn't supported (I'm assuming it's a hardware limitation in the case of this screen). 
- Straight up disable OTA updates for devices that do not support them in the app UI. I don't know what identifying data you get in the background, such that it could be automated with a safelist or the device self-reporting 

If this is the expected behavior, then this wouldn't be an issue. Let me know the etiquette you'd like me to follow here, and I can close it out, or you can leave it open as a potential task for the future.

ryanckulp, 2025-07-31:
thanks, i hear you. we'll adjust our `/api/display` endpoint to send back the "right build for the job" (device) soon. currently this information (device model) is not sent by the firmware to our server, so we handle those requests blind. this is partially why some BYOS/BYOD point to `/api/current_screen` instead, which does not instruct the device to update FW.

the first step to achieving this went live ~yesterday:
https://usetrmnl.com/api/models

next we'd like the FW to be model-aware, sending one of the model IDs in every request:
https://github.com/usetrmnl/trmnl-firmware/issues/199

finally we can send back a specific firmware build, similar to what we offer here: https://usetrmnl.com/flash

if this approach makes sense, feel free to close this issue and rest assured we're on it.

mmcnama4, 2025-07-31:
Thanks for the detailed response. Looks like you're headed in the right direction.