usetrmnl/trmnl-firmware #205: Infinte firmware update loop

raymondschouten, 2025-08-01:
My device was just stuck in an infinite firmware update loop, causing the device to continuously refresh the screen.

It displayed "firmware update available" and "firmware update failed". I had to shut down the device and start it again to break this cycle. After that it installed the update correctly.

emackinnon1, 2025-08-01:
I believe this may be related to the problem I just encountered as well. 

I have a DIY device (waveshare-esp32-driver) and had OTA updates on. My device also got stuck in the same firmware update loop described above. I tried manually updating to `v1.16.0` via PlatformIO in VS Code but got stuck in another loop: upon successfully uploading the new version and then attempting to connect to the WiFi network "TRMNL" and kicking off the whole setup process described [here](https://help.usetrmnl.com/en/articles/9416306-how-to-set-up-a-new-device), the device seems to reset and returns to the "Connect to WiFi" screen again and again, even after triple checking my SSID and password. It does this even after cleaning the build and re-uploading `v1.16.0` again.

I eventually reverted back to `v1.15.12` to get the device working again, but

raymondschouten, 2025-08-02:
I have an official build device, so I have no clue on how to see or do anything about this loop.

nhoizey, 2025-08-13:
> I had the same boot loop. I manually flashed the latest firmware via https://usetrmnl.com/flash

I had the same issue, thanks for this! 🙏

closed 2025-08-31 by ryanckulp