usetrmnl/trmnl-firmware #209: Add support for Wi-Fi Easy Connect to enable smoother initial device setup

liutikas, 2025-08-04:
It would be awesome to make the set up easier for novice users. If we set up Wi-Fi Easy connect it would make initial wifi credential management step a lot smoother.

There is an already existing ESP32 library to set this up:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_dpp.html

jrr, 2025-08-05:
This looks promising, and I look forward to a future where IOT devices don't have to implement captive portals.

But today, how is compatibility? It looks like DPP is a lower-level protocol that requires specific support by the client device. Can the last few versions of iOS and Android do it? Apart from mobile, is there a way to type a code into a desktop computer browser?

Another challenge is that today, the trmnl captive portal does more than just WiFi creds. (you can use it to configure a custom API hostname, perform a soft reset, etc.) Is there a way for easy connect to expose additional config values?

liutikas, 2025-08-05:
I am not suggesting the existing captive portal could still be there as fallback