usetrmnl/trmnl-firmware #213: Firmware no longer works with the Waveshare driver

skorokithakis, 2025-08-06:
The firmware has stopped working with the Waveshare ESP32 driver. It boots up into the wifi screen, it sometimes loads one image when the wifi credentials are entered, and then promptly forgets the credentials and boots back into the "connect to wifi" screen again.

<details><summary>Boot log</summary>
<p>
I (393) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (393) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (400) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (409) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (419) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I: src/display.cpp [50]: dev module end
                                       I: src/bl.cpp [239]: Display TRMNL logo start
I: src/display.cpp [491]: show image for array
                                              I: src/display.cpp [492]: maximum_compatibility = 1

                                                                                                 I: src/display.cpp [540]: Display refresh start
                                                                                                                                                I: src/display.cpp [543]: src/display.cpp [543]: Forcing full refresh; desired refresh mode was: 2

I: src/display.cpp [552]: src/display.cpp [552]: EPD refresh mode: 0

I: src/display.cpp [561]: display refresh end
                                             I: src/bl.cpp [248]: Display TRMNL logo end
I: src/filesystem.cpp [21]: SPIFFS mounted
                                          I: src/bl.cpp [255]: Firmware version 1.6.3
                                                                                     I: src/bl.cpp [256]: Arduino version 2.0.17
                                                                                                                                I: src/bl.cpp [257]: ESP-IDF version 4.4.7
                                                                                                                                                                          I: src/filesystem.cpp [206]: Filesystem Usage: 17068/233681
                        I: src/filesystem.cpp [211]:   831  last.png
                                                                    I: src/filesystem.cpp [211]:   15463  current.png
                                                                                                                     I: src/bl.cpp [2264]: NVS Usage: 192/630 entries (30.48
                                                                                                                                                                            I (980) wifi:wifi driver task: 3ffc5cb0, prio:23, stack:6656, core=0
I (2665) wifi:wifi firmware version: 1fd20f4
I (2665) wifi:wifi certification version: v7.0
I (2666) wifi:config NVS flash: enabled
I (2666) wifi:config nano formating: disabled
I (2670) wifi:Init data frame dynamic rx buffer num: 32
I (2675) wifi:Init static rx mgmt buffer num: 5
I (2679) wifi:Init management short buffer num: 32
I (2684) wifi:Init dynamic tx buffer num: 32
I (2688) wifi:Init static rx buffer size: 1600
I (2692) wifi:Init static rx buffer num: 4
I (2696) wifi:Init dynamic rx buffer num: 32
I (2700) wifi_init: rx ba win: 6
I (2704) wifi_init: tcpip mbox: 32
I (2708) wifi_init: udp mbox: 6
I (2711) wifi_init: tcp mbox: 6
I (2715) wifi_init: tcp tx win: 5760
I (2719) wifi_init: tcp rx win: 5760
I (2724) wifi_init: tcp mss: 1440
I (2728) wifi_init: WiFi IRAM OP enabled
I (2732) wifi_init: WiFi RX IRAM OP enabled
I (2740) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (2826) wifi:mode : sta (08:f9:e0:e1:28:90)
I (2826) wifi:enable tsf
I (2827) wifi:Set ps type: 1

I: src/bl.cpp [305]: WiFi NOT saved
I: src/bl.cpp [307]: FW version 1.6.3
                                     I: src/display.cpp [805]: Free heap in display_show_msg - 110592
                                                                                                     I: src/display.cpp [806]: maximum_compatibility = 1

                                                                                                                                                        I: src/display.cpp [809]: Free heap after bbep.allocBuffer() - 63488
               I: src/display.cpp [814]: Display set to white
                                                             I: src/display.cpp [834]: Paint_NewImage
                                                                                                     I: src/display.cpp [835]: show image for array
                                                                                                                                                   I: src/display.cpp [878]: wifi connect case
                                                                                                                                                                                              I: src/display.cpp [905]: Start drawing...
                           I: src/display.cpp [913]: display
                                                            I (3101) wifi:mode : softAP (08:f9:e0:e1:28:91)
I (3102) wifi:Total power save buffer number: 16
I (3103) wifi:Init max length of beacon: 752/752
[... 446 lines cut ...]
I (3397) wifi:Init data frame dynamic rx buffer num: 32
I (3402) wifi:Init static rx mgmt buffer num: 5
I (3406) wifi:Init management short buffer num: 32
I (3411) wifi:Init dynamic tx buffer num: 32
I (3415) wifi:Init static rx buffer size: 1600
I (3419) wifi:Init static rx buffer num: 10
I (3423) wifi:Init dynamic rx buffer num: 32
I (3427) wifi_init: rx ba win: 6
I (3431) wifi_init: tcpip mbox: 32
I (3435) wifi_init: udp mbox: 6
I (3438) wifi_init: tcp mbox: 6
I (3442) wifi_init: tcp tx win: 5760
I (3446) wifi_init: tcp rx win: 5760
I (3451) wifi_init: tcp mss: 1440
I (3455) wifi_init: WiFi IRAM OP enabled
I (3459) wifi_init: WiFi RX IRAM OP enabled
I (3466) wifi:mode : softAP (08:f9:e0:e1:28:91)
I (3469) wifi:Total power save buffer number: 16
I (3473) wifi:Init max length of beacon: 752/752
I (3477) wifi:Init max length of beacon: 752/752
I (3585) wifi:mode : sta (08:f9:e0:e1:28:90) + softAP (08:f9:e0:e1:28:91)
I (3585) wifi:enable tsf
I (3588) wifi:Set ps type: 1
</p>
</details>

emackinnon1, 2025-08-07:
I am going to follow this thread, too. I have the same Waveshare ESP32 diy set up and have the same boot loop when updating to the latest version (1.16.x)

I mentioned what I encountered and tried here in another issue a few days ago: https://github.com/usetrmnl/trmnl-firmware/issues/205#issuecomment-3146019734

skorokithakis, 2025-08-07:
@emackinnon1 can you try to bisect with your device? I'm afraid I don't have mine with me...

skorokithakis, 2025-08-10:
Yeah, I think that's it. Looks like my log says "detected extra-long press", so probably adding those lines back in will fix it. Not sure why it was removed...

emackinnon1, 2025-08-10:
That works for me now too. Not sure what happened earlier.

I would like to have it working with the latest version, but `v1.6.5` also results in problems. The `button.cpp` [logic](https://github.com/usetrmnl/trmnl-firmware/blob/v1.6.5/src/button.cpp) looks pretty different now. Have you tried that version?

emackinnon1, 2025-08-10:
I am not 100% positive on this, but from what I can tell with some of the changes having to do with waveshare boards, it seems like there was an assumption that everyone with the waveshare diy board setups would have buttons like the official versions? I only have one that is plugged into a usb core with no battery... I wanted to keep things very bare bones. 

Maybe someone can confirm, I am not really sure why some of the changes were made.

skorokithakis, 2025-08-12:
The issue here is that the firmware now assumes a button that grounds a pin, so it doesn't work without a connection (the ESP32 keeps waking up).

skorokithakis, 2025-08-12:
@emackinnon1 you are correct, that was the offending commit. I reverted it and added an internal pullup to pin 27, and now it works.

Basically, you either add a pulldown resistor (and maybe a button) to pin 33 (or whichever pin you want), or you can use the internal pullup and change the wakeup code. I did the latter.

Here's my diff, if you want to use it:

```
diff --git a/include/config.h b/include/config.h
index e263717..05053a8 100644
--- a/include/config.h
+++ b/include/config.h
@@ -64,9 +64,8 @@ enum WIFI_CONNECT_RETRY_TIME // Time to sleep before trying to connect to the Wi
 #define PIN_INTERRUPT 2
 #define DEVICE_MODEL "og"
 #elif defined(BOARD_WAVESHARE_ESP32_DRIVER)
-#define PIN_INTERRUPT 33
+#define PIN_INTERRUPT 27
 #define DEVICE_MODEL "waveshare"
-#define FAKE_BATTERY_VOLTAGE
 #elif defined(BOARD_SEEED_XIAO_ESP32C3)
 #define DEVICE_MODEL "seeed_esp32c3"
 #define PIN_INTERRUPT 9
@@ -84,7 +83,7 @@ enum WIFI_CONNECT_RETRY_TIME // Time to sleep before trying to connect to the Wi
 #if defined(BOARD_XIAO_EPAPER_DISPLAY)
 #define PIN_BATTERY 1
 #else
-#define PIN_BATTERY 3
+#define PIN_BATTERY 35
 #endif
 
 // #define FAKE_BATTERY_VOLTAGE // Uncomment to report 4.2V instead of reading ADC
diff --git a/src/bl.cpp b/src/bl.cpp
index bb9c7e8..6297071 100644
--- a/src/bl.cpp
+++ b/src/bl.cpp
@@ -123,6 +123,12 @@ void bl_init(void)
   }
 #endif
 
+#if defined(BOARD_WAVESHARE_ESP32_DRIVER)
+  // Enable internal pullup for the interrupt pin on Waveshare board
+  pinMode(PIN_INTERRUPT, INPUT_PULLUP);
+  Log_info("Configured PIN_INTERRUPT (pin %d) with internal pullup", PIN_INTERRUPT);
+#endif
+
   wakeup_reason = esp_sleep_get_wakeup_cause();
 
   if (wakeup_reason == ESP_SLEEP_WAKEUP_GPIO || wakeup_reason == ESP_SLEEP_WAKEUP_EXT0 || wakeup_reason == ESP_SLEEP_WAKEUP_EXT1)
@@ -1852,7 +1858,7 @@ static void goToSleep(void)
   // Configure GPIO pin for wakeup
 #if CONFIG_IDF_TARGET_ESP32
   #define BUTTON_PIN_BITMASK(GPIO) (1ULL << GPIO)  // 2 ^ GPIO_NUMBER in hex
-  esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK(PIN_INTERRUPT), ESP_EXT1_WAKEUP_ALL_LOW);
+  esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK(PIN_INTERRUPT), ESP_EXT1_WAKEUP_ANY_HIGH);
 #elif CONFIG_IDF_TARGET_ESP32C3
   esp_deep_sleep_enable_gpio_wakeup(1 << PIN_INTERRUPT, ESP_GPIO_WAKEUP_GPIO_LOW);
 #elif CONFIG_IDF_TARGET_ESP32S3
```

Feel free to discard my battery pin changes if they don't match yours.

Flattermann78, 2025-08-17:
Yes, I applied the whole patch and I just checked again.

jshiell, 2026-01-17:
I've taken a brute force approach to this, given I don't have any buttons wired in, and just disabled the GPIO sleep entirely. So far, so good.

```c
diff --git a/src/bl.cpp b/src/bl.cpp
index f8cdf32..43fad3e 100644
--- a/src/bl.cpp
+++ b/src/bl.cpp
@@ -1871,7 +1871,9 @@ static void goToSleep(void)
   // Configure GPIO pin for wakeup
 #if CONFIG_IDF_TARGET_ESP32
   #define BUTTON_PIN_BITMASK(GPIO) (1ULL << GPIO)  // 2 ^ GPIO_NUMBER in hex
+  #ifndef BOARD_WAVESHARE_ESP32_DRIVER
   esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK(PIN_INTERRUPT), ESP_EXT1_WAKEUP_ALL_LOW);
+  #endif
 #elif CONFIG_IDF_TARGET_ESP32C3
   esp_deep_sleep_enable_gpio_wakeup(1 << PIN_INTERRUPT, ESP_GPIO_WAKEUP_GPIO_LOW);
 #elif CONFIG_IDF_TARGET_ESP32S3
```

SBajonczak, 2026-01-28:
So i read These threads here. I own the esp23 waveshare Board. It fruststung the it loop i to a Screen than into the startscreen  then in the next image...but without pause.

So as I read there is a button identify issue then? So as I identified in the logs on the device the sleep command will also not set. I think that there are two kond of problems here or i am wrong?

jshiell, 2026-01-28:
Update: while the hack above did get my device working, I ran into constant refresh errors on the screen, with parts being faded. I've reverted back to 1.5.12 and it's working normally again. So it may be that this isn't the only change that's affected this board.

skorokithakis, 2026-01-28:
You *may* be able to fix this by forcing full refresh by checking "maximum compatibility" in the settings here:

https://trmnl.com/devices/11212/firmware/edit

I'm trying it now and will see.