usetrmnl/trmnl-firmware #220: Using certain images inverts colours

Matt-Mcl, 2025-08-15:
I'm using the latest firmware at the time of writing (v1.6.5) and I'm getting a strange issue where the whole display is inverted in colour when displaying certain images.

Not sure what version has caused this but it didn't seem to affect be previously on 1.5.x.

Below is a bitmap that, I believe, is very clearly a white background with a cute outline of a frog in black.

[image]

Here is the output you see on the trmnl.

[image]

My trmnl always seems to render this inverted and I cannot figure out why. If anyone else could test getting this bmp onto their trmnl to see if there is something strange going on in the firmware to invert the colours.

Matt-Mcl, 2025-08-16:
Thank you for the great explanation!

For more context I'm currently using the byos_django server with a few modifications. 

However, I haven't altered the way it generates the screens. It currently uses playwright to take an html page and convert it to the bmp images. If I were to just change this to be png would this resolve my issue?

Then is there a way to set the palette to be 4 bit grayscale?

I'll have to have a look into this later. 

Thanks again!

delhoume, 2025-08-16:
> Then is there a way to set the palette to be 4 bit grayscale?

You dont't want to use palette PNG at all

 The TRMNL page https://docs.usetrmnl.com/go/imagemagick-guide explains how to generate images.

It should be updated for 1 bit PNG  to match what is done for 2 bits, as the principle is the same
you generate a grey palette of N bits (image type can be grey, palete, or truecolor , all theat matter are the colors in the data
you use this palette to remap the colors
you save in preferred / supported format greyscale format
-> your are guaranteed they will display as expected.

1 bit PNG
```
magick -size 2x1  xc:#000 xc:#fff +append  -type Palette /tmp/colormap-1bit.png
magick input.png -resize 800x480\! -dither FloydSteinberg -remap colormap-1bit.png -define png:bit-depth=1 -define png:color-type=0 output1bit.png

```

2 bit PNG
```
magick -size 4x1  xc:#000 xc:#555 xc:#aaa xc:#fff +append  -type Palette /tmp/colormap-2bit.png
magick input.png -resize 800x480\! -dither FloydSteinberg -remap colormap-2bit.png -define png:bit-depth=2 -define png:color-type=0 output2bit.png
```

4 bit PNG (future X panel)
```
magick -size 16x1 xc:#000 xc:#111 xc:#222 xc:#333 xc:#444 xc:#555 xc:#666 xc:#777 xc:#888 xc:#999 xc:#aaa xc:#bbb xc:#ccc xc:#ddd xc:#eee xc:#fff +append  -type Palette /tmp/colormap-4bit.png
magick input.png -resize 800x480\! -dither magick input.jpeg -resize 800x480\! -dither FloydSteinberg -remap colormap-4bit.png -define png:bit-depth=4 -define png:color-type=0 output4bit.png
```

like for the frog  BMP, you can generate paletted PNGs, for example 2 bit, that will render visually the same as grayscale on a PC viewer, but may not on OG TRMLN, because you have no guarantee on palette order (could be White, Grey2, Black, Grey1 or whatever combination) ad TRMNL FW does not remap the colors (needs to go through all data)

`magick input.png -resize 800x480\! -dither FloydSteinberg -remap colormap-2bit.png -define png:bit-depth=2 -define png:color-type=3 output2bit_palette_if_you±_are_luckyit_displays_fine_if_not_colors_may_be_swapped.png`

You can use the "Alias" plugin to try and display 2bit PNGs